Denvi/Candle #3: Error at start-up with Windows XP

jrhansen, 2015-09-06:
Hi there!

I just saw some screen shots of grblControl, and it looks great! Unfortunately, I am unable to start the program, because of the following error:

-----
The procedure entry point vsprintf_s could not be located in the dynamic link library msvcrt.dll
-----

I am trying to use grblControl on my old laptop running Windows XP (if that matters). Can you help?

Thanks in advance
jrhansen :-)

Denvi, 2015-09-06:
Need to rebuild with msvc compiler and test on winxp. It will take some time.

jrhansen, 2015-09-06:
Thanks for the quick reply. Your help is highly appreciated!

Let me know if you need more info, or if I can be of any other help.

Denvi, 2015-09-06:
Builded with QT 5.3.1 MinGW 4.8.2 32bit (there were much more problems with MSVC compiler, than I assumed).
Some visual artefacts with text overpainting over opengl window for now.
Download link: [grblControl_0.5.2_xp.zip](https://github.com/Denvi/grblControl/releases/download/0.5.2/grblControl_0.5.2_xp.zip)

jrhansen, 2015-09-06:
The new build seems to work! I have not yet had the PC connected to the CNC, but I can start up the application without any errors, and I don't even see any of the artifacts that you mentioned. Maybe that will change, when I start using it for real.

Thanks again for an extremely quick response! :-)

closed 2015-09-14 by Denvi