Denvi/Candle #5: Prebuilt binary fails to launch on Ubuntu 14.04

powercycle, 2015-09-20:
Tried downloading the pre-built binary and it fails at command line execution.

There is an Error, but the message that comes after doesn't show anything intelligible.

Also, tried cloning the repo to build it myself, but no make files.

How should I build this codebase on my Ubuntu machine?

Thanks for your guidance.

Denvi, 2015-09-20:
Hello, can you copy/paste console messages before error?
To build from source code you need install Qt5 (latest version recommended, >=5.3 needed to use serial port) on your machine (https://wiki.qt.io/Install_Qt_5_on_Ubuntu). 
Then in command line:
1) cd <path_to_source_code>
2) qmake
3) make

powercycle, 2015-09-20:
Here is the output when trying to run the grblControl binary.
[image]

powercycle, 2015-09-22:
I apologize..I did not mean to close this issue.

Denvi, 2015-10-03:
Fixed in version 0.6.

powercycle, 2015-10-05:
I'll give it a shot this week.

powercycle, 2015-10-06:
This worked great!   Thanks!

closed 2015-10-06 by powercycle