Denvi/Candle #6: Spanish Translation

lucasvass, 2015-10-01:
Hi there!
i´m currently working with foam cutting machines,grbl is perfect for this kind of aplications and so it grbl Control, but for some people around here,english is still a limited subject.
Is there any chance to get any Xml so i can be able to translate it to spanish?
Thanks in advance!

Denvi, 2015-10-02:
Hello, "xml" file here: [grblControl_es.ts] (https://github.com/Denvi/grblControl/tree/master/translations/grblControl_es.ts) (it`is actual version of latest source code). 
You can translate it with "Linguist" app from Qt or with regular text editor.
Below is a sample message translation.

Original message:
```xml
    <message>
        <location filename="../gcodetablemodel.cpp" line="17"/>
        <source>Command</source>
        <translation type="unfinished"></translation>
    </message>
```
Translated:
```xml
    <message>
        <location filename="../gcodetablemodel.cpp" line="17"/>
        <source>Command</source>
        <translation>Comando</translation>
    </message>
```

After .ts file translation you can do pull request or just send file some another way to me.
I will add translation to next build.

Denvi, 2015-10-03:
Added to version 0.6.

closed 2015-10-30 by Denvi