Denvi/Candle #8: a few small requests

pihnat, 2015-10-21:
Hi,
Great piece of software, so easy to use. Well done!

Just a few small things.
1. In the G-code display box in the 'State' column you use the word 'Sended' instead of 'Sent'.
2. In the file-open popup box there are 3 types of files, *.nc, *.ncc and *.tap. Could you also include *.txt?
3. When the mill is running it seems the only way to pause the machine is to click the 'Pause' button on the screen. Would it be possible to implement a keyboard press as well, for example spacebar? There's no problem with the onscreen Pause button but sometimes when working around the mill it would be easier to just tap spacebar to pause the machine and then tap it again to continue.
4. I noticed that sometimes the 3D picture isn't drawn. I tried a few things and found that the G00 line near the start of a program MUST have all three parameters X, Y and Z. For example if I use G00 X0 Y0 nothing is displayed. As soon as I change this to G00 X0 Y0 Z2 the whole 3D image appears. Is this intentional?

regards
Peter

Denvi, 2015-10-21:
Hello.
1. Thanks, it is my mistake, will be fixed.
2. Ok, will be added.
3. There are problems with form controls focus (you can press 'Space' in command textbox for example to enter a space but not to pause work). So i can only add more complex shortcut for now. Actually, I think the CNC shall have a hardware 'Pause' button.
4. It is intentional, but it will probably be corrected in future versions. The beginning of the first line segment will be the current position of the tool.

pihnat, 2015-10-21:
Yes, I understand the problem with form controls focus so don't worry about trying to implement a way around it. At least there's a Pause and Reset button on the screen and we can implement a hardware stop/pause button.

thanks
Peter

Denvi, 2015-10-30:
1. Fixed
2. Added
3. "Pause" button gets focus on "Send" button press, can be pressed with "Space". Solved partially.

closed 2015-11-08 by pihnat