Denvi/Candle #10: Restore X,Y Z and Reset

pihnat, 2015-10-23:
Hi Denvi,
1. When you press the 'Restore X, Y, Z' button only X and Y are restored. Z should move too.

2. Pressing 'Zero XY' and 'Zero Z' works OK to offset from Machine co-ords. But if you then press 'Reset' to reset the Arduino the current Work co-ords are replaced by the Machine co-ords. Shouldn't the Work co-ords stay the same?
This becomes a problem as follows: let's say you offset & then use 'Zero XY' and 'Zero Z' buttons to set a new 0,0,0 and then send the G-code program. All is OK until you want to quit part way through the program. Pressing and releasing 'Pause' works correctly but when paused if you then press the 'Reset' button to quit sending G-code the Work co-ords become the same as the Machine co-ords and you lose the current Work co-ords.

regards
Peter

Denvi, 2015-10-23:
Work coordinates become the same as machine after reset due to the properties of the GRBL firmware.
It`s a part of reset procedure.
'Zero XY' and 'Zero Z' buttons both stores work coordinates offsets (and also saves thems to settings file to restore on program restart).
'Restore XYZ' restores all 3 offsets, but move tool only on XY plane to prevent tool collisions.
So after reset you can restore your work offsets.
I think It would be more appropriate to separate the functions of restoring XYZ offsets and move tool to zero XY (Z). But I did not encounter cases where you need to move the tool along the axis z to 0. Usually need only XY.

pihnat, 2015-10-25:
I will think some more about the Restore XYZ function.

The other point to do with Reset: yes I see now when grbl gets the cntl-x command it resets and makes the work co-ords = machine-co-ords.
In grblControl when you press Send the only 2 options then are Pause or Reset grbl. What's missing is a Stop button. There are many times when I do a test run but want to stop part way through or if I'm showing the software to others I want to stop after a few lines. Unfortunately the only way to leave the Send is to press Reset which also resets the Arduino. A Stop button would be very useful because you could quit from Send and it would leave the Work co-ords at their current values.

Denvi, 2015-10-25:
There is one caveat to the implementation of the "Stop" button. The GRBL-firmware has a movements buffer & commands buffer, interrupt program send will not lead to an immediate stopping of the machine.

pihnat, 2015-10-25:
I was surprised that grblControl has an immediate stop when using the Pause button, no one else's program does that, they all keep going until buffer is empty. Your program pauses immediately.
The Stop button (or maybe call it Quit) should stop just like the Pause button does but the main idea is to leave the Sending function. At the moment once you have pressed Send you can't get out of Send mode without pressing Reset which resets the Arduino. There should be another button which quits Send mode but doesn't reset the Arduino.

Denvi, 2015-10-25:
Ok, with current GRBL-firmware it seems to be imposible.
Firmware supports immediate pause with '!' command, but there is no command to immediate stop in firmware besides reset command ('ctrl+x').

pihnat, 2015-10-25:
OK I was just reading about grbl. So you must use a Feed Hold command (!) to pause it and there isn't any other command to clear the buffer or stop except for cntl-x reset. So I understand why the program would need to restore work co-ords, etc after cntl-x.

pihnat, 2015-10-25:
I'm not so concerned about emergency stop because a person has 2
    options - they can press the Reset button at top right which sends
    cntl-x or they can implement a hardware emergency stop button
    connected directly to the Arduino. In these cases the red Alarm
    indicator will come on because co-ords will no longer be accurate.
    
    I'm more interested in a way to quit 'Send' mode.
    You have 4 buttons: Open, Reset, Send and Pause. It would be nice to
    have a Quit button too which is enabled after Send is pressed.
    Operation would be as follows:
    - after you press Send button the G-code is being sent to the grbl
    controller
    - if you press Pause the CNC is paused. Releasing Pause resumes the
    CNC
    - pressing Quit will do a controlled stop as follows
      a. if Pause was already pressed then Quit will send cntl-x and
    then restore Machine co-ords
      b. if not paused when Quit is pressed it will send Pause command
    (!), then cntl-x and then restore Machine co-ords.
    
    This way I think it should be possible to leave the sending function
    in a controlled way and not lose Machine co-ords.
    regards
    Peter
    
    Denvi wrote:
    
    
      
      Ok, with current GRBL-firmware it seems to be imposible.
        Firmware supports immediate pause with '!' command, but there is
        no command to immediate stop in firmware besides reset command
        ('ctrl+x').
      —
          it on GitHub.

pihnat, 2015-10-26:
With regards to the main Reset button at upper right, when pressed the program should read current Work co-ords first then send cntl-x.
a. If return message from grbl is Alarm then grlControl should act the same as it does now
b. if current positions are not lost then Work co-ords on the screen should be updated to ones just read.

Denvi, 2015-10-26:
'Reset' is a 'Reset' and it should reset GRBL state including work coordinates offsets.

pihnat, 2015-10-27:
Yes, I agree that an overall Reset should put grbl into its initial state. But this should only be required on few occasions for example between jobs, after an emergency shutdown, etc.
I think it's more important not to lose Work co-ords. When setting up for a job it can take quite a bit of time positioning the tool and zeroing axes for a job. Many times this involves using a feeler gauge to precisely space the tool from the workpiece, etc.
And then in one press of Reset all that info is lost and you have to re-start the calibration process again.

Denvi, 2015-10-27:
Don`t push "Reset" button if you don`t want to lose "that info". Anyway you can just press "Restore XYZ" button after reset.
"Abort" button added to last commit. Binary file for windows: (grblControl.exe)[https://github.com/Denvi/grblControl/blob/master/bin/grblControl.exe]

pihnat, 2015-10-27:
Fantastic work, thank you.
After pressing Pause I take note of the Work co-ords and then press Abort. 99.9% of the time the Work co-ords stay the same but on the odd occasion the X and Y values will change by some amount. I'm not sure what conditions this happens under & I'm not sure if it's a random amount or the same offset each time. It can be up to 1mm difference.
The tool in the visualisation does move to the different Work co-ords but the motors don't move. I will keep experimenting to find out what conditions it happens under and how much the offset is. It does not happen very often so is a bit hard to duplicate.
Thanks again

pihnat, 2015-10-28:
The offset seems to be either the Restore XYZ values or the stored G54 values, I'm not sure which. It still seems to be random but only after pressing Pause and then pressing Abort. Most of the time there's no problem but just occasionally the offset is added to the Work co-ords for no apparent reason.

Denvi, 2015-10-28:
"Zero XY", "Zero Z", "Restore XYZ" buttons works with G92 command.
"Abort" button also restores only G92 offsets for now.
So, it may be problem with G54-G59 offsets. Do you use them?

pihnat, 2015-10-28:
No I don't use them but I noticed only G54 is set to some values. Does your program change them any time? If not then I will clear them & see what happens.
It seems strange that the problem only occurs at random, there doesn't seem to be any pattern.

Denvi, 2015-10-28:
Program doesn't changes G54-G59 offsets.
May be G-codes (g-code program), you using, includes G10 L2, G10 L20 commands?

pihnat, 2015-10-28:
No, only the simple ones G0, G1, G2, G3, G18, G21 & G90.
Yesterday the Abort button only played up twice but on the second occasion the Working co-ords became the same as Machine Co-ords which was a large difference in position. I will continue checking it today to find a pattern.

pihnat, 2015-10-28:
OK I can make it play up each time as follows:
- G54, etc parameters in Arduino are all 0
- I edit grblControl "settings.ini" file so stored X, Y & Z are 1, 2 & 3 respectively
- start grblControl
Work co-ord are 0, 0, 0
Mach co-ord are 0, 0, 0
- load a G-code file
- press Reset and then Send
- a few lines into the program during a G01 command press Pause
Work co-ord are -0.058, -1.203, -3.000
Mach co-ord are -0.058, -1.203, -3.000
- now press Abort
Work co-ord now become -1.058, -3.203, -6.000
Mach co-ord are -0.058, -1.203, -3.000

So the Restore XYZ values have been SUBTRACTED from the Work co-ords.
The mill doesn't move to the different Work co-ords but the tool in the 3D visualisation has moved.

Denvi, 2015-10-29:
Fixed in last commit.

pihnat, 2015-10-29:
Fantastic, it works now.
Sometimes the Work co-ords can change by 0.001mm between pressing Pause and Abort. I assume this would just be some rounding error because of the subtraction maybe?

Denvi, 2015-10-29:
> Sometimes the Work co-ords can change by 0.001 or 0.002 mm between pressing Pause and Abort

How about machine coordinates? They remain invariable?

pihnat, 2015-10-29:
Machine co-ords stay fixed, only Work co-ords change.

Denvi, 2015-10-30:
Meanwhile I couldn't reproduce similar behavior.
I will test.

pihnat, 2015-11-01:
It may only be with G02/03 command, I need to test more. At the moment this will show it:
G90
G21 
G17
G00 X2 Y0 Z0
F80
G01 Z-3 
G02 X-2 Y0 I-2 J0
M30

When you are about halfway round the curve press Pause, check Work co-ords then press Abort. It doesn't do it every time but seems to be at random. But the error accumulates each time it happens.

For example when I pressed Pause I got
WC  0.255  -1.983  -3.000
MC  0.255  -1.983  -3.000
then when I press Abort the Work co-ord for Y changed to -1.984, all others unchanged.

I will keep checking to see if it happens besides G02/03.

Denvi, 2015-11-01:
Finally.
This mistake is appears when using a microstep. For example 800 steps on one millimeter.

>  I assume this would just be some rounding error because of the subtraction maybe?

Most likely indeed.
And the only decision is the restoring offsets with "G92.3" command. Which isn't supported by GRBL yet.

pihnat, 2015-11-01:
I'm using 400 steps/mm so I can see there would be some rounding errors.

pihnat, 2015-11-05:
Re Abort button operation:

While sending a file I can press Pause and then Abort and it always stops without a problem.
However while sending a file if I just press Abort without the Pause it works most times but sometimes I get Alarm condition. I understand that grblControl sends the Feed Hold command followed by cntl-X to implement the Abort but I'm guessing that the tool is probably still decelerating when the cntl-X is sent & that's why Alarm is raised. Would that be the cause?

pihnat, 2015-11-07:
Following up:
I re-checked grbl wiki & see the problem. When a Feed Hold command is sent to grbl and you check Status it returns 'Hold' when it starts deceleration right through the stop until you resume the program. To me this is confusing because the Status should be 'Hold' only AFTER the feed has actually stopped.

I'll check previous issues in grbl and if not mentioned I should raise it. What do you think?
If fixed it would make the Abort function in grblControl or any other control software much more robust.

Denvi, 2015-11-07:
> Next major version of grbl will have short confirmation messages for real time commands.

Glad to see.

Anyway, i have increased delay between getting "Hold" status & sending "Ctrl-X" command to 500 ms.
I will try to find another solution, until next major grbl version.

pihnat, 2015-11-08:
Works much better now, a great solution. Interesting to see what confirmation messages they implement in the next version of grbl.

thanks again
Peter

closed 2016-01-13 by Denvi

Alex535nvx, 2020-03-01:
Hi. I have bad problem with all grbl controller version and also candle version
I can't open the grbl software. When I click to opening my computer has error (windows was unable to open file).
Can you help me why I have this problem. I have a laptop but in my laptop I don't have this problem.

Alex535nvx, 2020-03-01:
> Hi. I have bad problem with all grbl controller version and also candle version
> I can't open the grbl software. When I click to opening my computer has error (windows was unable to open file).
> Can you help me why I have this problem. I have a laptop but in my laptop I don't have this problem.

And only I just can open grbl 0.5.2
Befor that I have problem with this error of windows(window has missing the Qt5core. dll). Then I give Qt5core and put on my windows the problem was solved but after that I had windows error message with (windows was unable to run ) 

I can be for my windows or some Prerequisites library or software to opening file. Because my I have install new window
My window in both laptop and computer is 7. 64bit

pihnat, 2020-03-19:
Hi Alex535nvx
Go to the Candle page & about 3/4 way down the page download the Windows version of Candle. Currently it's version 1.1.7. You have a problem because I think you haven't unzipped it (you can't run Candle from inside a zip file).

So double click the   candle_1.1.7.zip     file that you downloaded and you'll see a folder called  Candle in the window. **Drag** that folder onto the desktop (it will be unzipped automatically).

Inside that folder you will see all the libraries, etc and you will see a file called   Candle.exe
Double click that file and it will run.

Let me know if that solves your problem.

Alex535nvx, 2020-03-27:
Thanks pihnat for your answer :
I did this method for solving my problem. But I still have problem with that.

pihnat, 2020-03-28:
I don't understand why you still have the problem. In the Candle folder that you unzipped you will see many .dll files including Qt5Core.dll so everything that Candle needs to run is already in that folder.
I have run Candle on both win7 (64 bit) and win10 without any problems. Very strange.

pihnat, 2023-03-12:
Hi Peter,
This thread is closed. You should open your query as a new issue since it's not related to the previous discussion and please explain a bit more - it's not obvious why the tool should break, you need to explain what's happening in more detail.
regards