Denvi/Candle #9: XZ curve & M30 issues

pihnat, 2015-10-23:
Hi,
I've discovered a couple more small quirks with grblControl:
1. M30 when executed should move the highlighted G-code line back to the first line
2. When creating an arc in the XZ plane the Arduino performs the correct movement with the tool however the visualiser draws a straight line instead of an arc. Try this:
G90
G18
G21
F200
G0 X0 Y0 Z0
G1 X5
G3 X5 Z10 I0 K5
M30

If you replace the G3 line with
G3 X5 Z10 R5
the motion is also correct except the visualiser doesn't display any arc or line.

kind regards
Peter

Denvi, 2015-10-23:
Now, as the arc axis is only supported axis Z. Others will be added in future (G17, G18, G19 codes).

pihnat, 2015-10-25:
The reason I am asking about arc support in M18 (xz) mode is because I have tried grblControl with a lathe. It works very well except the 3D pic is vertical instead of horizontal because lathes use X and Z. However the path of the tool is correct except for the arc.
Do you have any plans to one day implement a lathe mode? It could be a setting where you select mill or lathe view.

Denvi, 2015-10-28:
>Do you have any plans to one day implement a lathe mode? It could be a setting where you select mill or lathe view.

At the moment no such plans.

Denvi, 2015-10-30:
G17, G18, G19 commands support added.
Tested with following program:
```
G90
G18
G21
F200
G0 X0 Y0 Z0
G1 X5
G3 X5 Z10 I0 K5
G19
G3 X5 Z0 I0 K-5
G17
G3 X0 Z0 I-2.5 K0
M30
```

Denvi, 2015-10-31:
Processing of the M2 & M30 commands is fixed.

pihnat, 2015-11-01:
Still a slight problem with G18. Try the following
G90
G21
F200
G18
G00 X0 Y0 Z0
G01 X7 Z0
G03 X7 Z-20 I-7.0 K-10
G01 X7 Z-24
G01 X14 Z-24
M30

The  arc is drawn in-correctly but when Sending the tool does trace the correct path in the 3D view.

Denvi, 2015-11-02:
Radius calculation problem. Fixed.

pihnat, 2015-11-02:
I just downloaded it and it works perfectly. Fantastic job with the whole project!!
thanks again
Peter

Denvi, 2015-11-03:
Meanwhile I commited the intermediate version. I will test more.

pihnat, 2015-11-03:
It's working now, xy and xz arcs are good. I will keep testing too.

closed 2015-11-08 by pihnat