Denvi/Candle #11: French translation

Xav83130, 2015-10-27:
Hi,
I did the translation in French, but I do not know how to push it on the Git.
I have not your email address to send it.
can you help me ?

Denvi, 2015-10-28:
Just press green button left to "branch" selector in Your forked repository, button has tooltip: "Compare, review, create a pull request".
Then press "Create pull request" button.

Xav83130, 2015-10-28:
I think it's good?
grbl Control is really simple and effective good job.
Thank you for sharing.

Denvi, 2015-10-28:
Don't see any pull request.

Denvi, 2015-10-29:
Added. Thank you.
You can test it with last binary file for windows from "bin" folder & "grblControl_fr.qm", "qt_fr.qm" files copied to "translations" directory.

Xav83130, 2015-10-29:
with pleasure,
I am under linux, and I'm not very good for "compiling".
I am looking forward that you compile the linux version for testing.

Denvi, 2015-10-29:
Check linux binary in "bin" directory: [grblControl](https://github.com/Denvi/grblControl/blob/master/bin/grblControl)

Xav83130, 2015-10-29:
ok I've experienced in my work with windows.
I saw a few small mistakes.
I correct in the evening.

Xav83130, 2015-10-29:
It seems to me better.
Sorry for the inconvenience.
See you soon

Xav83130, 2015-10-31:
Hi,
Translation look pretty good, I transmits the info on some French forum.
to see if there are bugs.
nothing to see, but would it be to you to have the updated version of linux with the latest changes more regularly?
thank you very much for your work.

Xav83130, 2015-11-02:
Hello,
I am looking for the meaning of "Zb" and "Zt"?
if someone an idea.

Denvi, 2015-11-03:
Zb - Z-bottom coordinate for heightmap probing;
Zt - Z-top.
Also one more new line is added to the program. Check *.ts file please. Thanks.

Xav83130, 2015-11-10:
Ok it looks good.

Xav83130, 2015-12-01:
Hi,
For this, "(used in visualizer and heightmap modification procedure)"

I understand "used in visualizer and heightmap"  but i don't understand this "modification procedure" if you explain?

Denvi, 2015-12-02:
This refers to the procedure of converting a file using the height map.
I rewrite this string later in a more understandable form.

Xav83130, 2015-12-05:
I added the translation, but I can not send, permissions problem with git.

Denvi, 2015-12-12:
Hello, explain, what the exact problem is?

Xav83130, 2015-12-12:
When i commit with GitHub i have this message.
With other Git i don't have problem.
[image]

Xav83130, 2015-12-12:
I'm really bad with Git
I succeeded but not without difficulty.
I especially hope not to do anything.
A ++

closed 2015-12-12 by Xav83130