Denvi/Candle #16: a few questions

pihnat, 2015-11-04:
1. I've got grblControl working with just over 36,000 lines of G-code. What's the limit to the number of lines it can handle?

2. In 'Service Settings' what does the 'Command auto-completion' option do?

3. The Feed Override setting doesn't change the feed rate while G-code is being sent to the Arduino. In other words it doesn't change the feedrate in real time but appears to only get read by grblControl when the Send button is first pressed, is that right?

thanks in advance
Peter

Denvi, 2015-11-04:
1. With "Geometry simplify" option checked and "Autoscroll" option unchecked I think limit is about several million lines (depends on RAM amount). Otherwise it is about 100k lines.
2. Enables text completion feature of command input field in the "Console" panel. Input field can store up to 5 last commands. Feature uses that last commands for completion of the currently input command.
3. Not certainly in that way, it overrides feedrate of all the commands will be sent after you set/change "Feed override" option. So you can change feed while sending a file, but there will be a delay due to commands buffer.

pihnat, 2015-11-05:
re 3.
Am I right in saying that no matter what you set the override to grbl won't exceed the max speed set in its $110, $111 and $112 settings?

BTW currently there's no change to feed rate if you change the override value while a file is being sent. It does work perfectly if you change it before you send the file.

Denvi, 2015-11-05:
Program doesn't care about max speed set of GRBL, but, as i know, firmware keeps track of feed, it does not exceed the maximum values.

pihnat, 2015-11-05:
Works well! Thanks for the prompt attention.
BTW the exe is now only 655K in size instead of over 2M. What happened?

Denvi, 2015-11-06:
Fonts resources removed to reduce commits sizes.

closed 2015-11-08 by pihnat