Denvi/Candle #18: Jog positioning

Xav83130, 2015-11-19:
Hi,
In full screen, the buttons of the manual control (Jog) are partially hidden.
would it be possible to improve this?
here is a picture: http://pix.slic.it/p/5hr
Thanks

pihnat, 2015-11-20:
You may have to use a smaller font size or increase screen resolution.
On a laptop with 1366x768 screen I start losing jog buttons with font size of 10 or higher but all is OK with 8 & 9 font size.
On my desktop computer with 1680x1050 screen there's no problem with font size of 12 - all jog buttons are fully visible.

Denvi, 2015-11-20:
I can decrease console window minimum height:

[image]

Xav83130, 2015-11-21:
may be given the opportunity to increase / decrease the console window:
[image]

Denvi, 2015-11-27:
Commited this:

[image]

Xav83130, 2015-11-27:
Perfect! I look forward to testing.
Thx

Denvi, 2015-11-30:
Added binaries for flicker-free version:

[image]

Xav83130, 2015-11-30:
Even better ... thank you.
I regret not knowing compile .... to enjoy this change. ;-)

Denvi, 2015-12-01:
Just copy file from binaries to program folder:

[Binaries here](https://github.com/Denvi/grblControl/tree/master/bin)

linux: [grblControl](https://github.com/Denvi/grblControl/tree/master/bin/grblControl)
windows: [grblControl.exe](https://github.com/Denvi/grblControl/tree/master/bin/grblControl.exe)

Xav83130, 2015-12-05:
Ok I tested, I find it very well.
Thank you

closed 2015-12-05 by Xav83130