Denvi/Candle #19: Compiling problem

losmichal, 2015-11-28:
Hi

I'm try to compile this in Qt but still geting lot of error can you please tell me what compiler you are using. I try VC++8 and VC++14 but with lot of error 
[image]

Denvi, 2015-11-28:
Hi, i'm using MinGW compiler.

losmichal, 2015-11-29:
Hi

Thank you for quick replay.
I have add MinGW 4.9.2 32bit and it has compile with lot of Warning, it's run but with about Visualization 
Can you please help me with this ??

[image]
[image]

Denvi, 2015-11-29:
It is possible not to pay attention to warnings, it's just about unused variables.
Visualizer & table have to be adjusted.
Visualizer can be fixed with "Set to defaults" button on settings form.
Table - with proper settings file (you can copy "settings.ini" file from last release archive to program build folder).

losmichal, 2015-11-29:
Thank You it's work :)
I'm going to translate it, do i have to change all label name etc. or i can add another language

Denvi, 2015-11-29:
1) In .pro file declare your translation in "TRANSLATIONS += ..." string (use ISO 639 for name suffix).
2) Create .ts file with "lupdate" command from "Tools > External > Linguist" menu.
3) Make translation by editing .ts file with "Linguist" app or other text editor.
4) Compile .qm translation file with "lrelease" command from "Tools > External > Linguist" menu in "Qt creator" app.
5) Copy .qm file to "translations" directory of "grblControl" app build/run folder.

closed 2015-12-24 by Denvi