Denvi/Candle #20: Saving the adjusted gcode

Super169, 2015-12-01:
Thanks for sharing the software, really excellent.

I like the feature of heightmap very much, it solve the problem of leveling after re-assembly.  Thank you very much.

However, when I try to save the gcode, it will save the original version instead of the adjusted version.  

As I have built an off-line milling tools with my CNC, so that it can use the file in SD card without computer.  May I know if it's possible to save the adjusted gcode to a file, so that it can be used for off-line milling.

Thanks in advance.

Denvi, 2015-12-01:
There is no such function for now.
Any offers how to insert this functionality in the user interface of the program?

Super169, 2015-12-02:
Thanks for your prompt response.

However, I'm sorry that I don't know how I can use the exe file in  iss20 branch binaries.
I have downloaded the file, it has only 38KB (compare to the one in v0.7 has 657KB), and the following error is prompted when I try to replace the exe and execute it.

"The version of this file is not compatible with the version of Windows you're running.  Check your computer's system information to see whether you need an x86 (32-bit) or x64 (64-bit) version of the program, and then contact the software publisher."

I'm using a 64-bit version Windows, does it mean that it can only run on 32-bit  OS?  But there has no problem running v0.7.

Thanks a lot.

Denvi, 2015-12-02:
I think, you saved .html.
"exe" file have 666 kB length.

[image]

Super169, 2015-12-02:
I just find another issue that M05 has been removed in the transformed code.
So the motor will not stop after milling.

However, if I make some modification on the code list, then check "use heightmap", it will not remove those M05.

Is it intends to remove those M05 from gcode file?

Denvi, 2015-12-02:
It's a bug, fixed day ago, but need to be tested.

Super169, 2015-12-02:
oh, i see.  I can download the file now.  I just right click "grblControl.exe" and save it, so it only download a 38KB file.  Thanks a lot.  

I can save the transformed gcode to file now. ^_^

closed 2015-12-02 by Super169