Denvi/Candle #21: Can it use the measured value directly to the heightmap?

Super169, 2015-12-02:
I just try to build my first PCB using grblControl with heightmap.  However, it hasn't milled on the PCB, it just traveled on the air.

I checked the transformed gcode, and study the probe action again.  It seems that it will first measure the height of 0,0, and then use the z value for adjustment on the subsequent measurement.  For example, if in 0,0 the z value is -0.3, then all subsequent measurement will be adjusted by 0.3.  So, the transformed z value in 0,0 is always 0.  As a result, it will mill at the position 0.3mm above the target position.

Since my DIY machine cannot recognize the home position, it will reset 0,0,0 to the position whenever the machine started.  If it can detect the level in 0,0, that will be great as I have no need to tune the z axis in 0,0 before milling.

And now, it has already measured the difference in all position including 0,0, but it will reduce the value by the measurement in 0,0.  So, I have to tune the z position in 0,0 before starting the machine.

Is it possible to use those measured z-value directly without offset the measurement in 0,0?

Thanks a lot.

Super169, 2015-12-02:
For more detail, what I have done now is to turn on the "Show G-code program commands" option un Console, then update the table of heightmap using the value displayed in console.

For example, if the following result show in the console,

G0X0Y0 < ok
G38.2Z-1 < [PRB:0.000,0.000,-0.429]
ok
G0Z1 < ok
G0X0.000Y0.000 < ok
G38.2Z-1 < [PRB:0.000,0.000,-0.429]
ok
G0Z1 < ok
G0X5.000Y0.000 < ok
G38.2Z-1 < [PRB:5.000,0.000,-0.385]
ok
G0Z1 < ok
G0X10.000Y0.000 < ok
G38.2Z-1 < [PRB:10.000,0.000,-0.342]
ok
:
:
I will go to change the value in 0,0 from 0.000 to -0.429
and the value in 5,0 from 0.044 to -0.385, the value in 10,0 from 0.087 to -0.342, .....

So, if it can use the value returned from GRBL directly, it can save much time for post-adjustment.

Denvi, 2015-12-02:
I will think about it.

For now, I'm using commands like following:
```
G91 G21
G38.2 Z-30 F75
G0 Z1
G38.2 Z-1 F7.5
G92 Z0
```
in (0,0) point to set Z-axis before milling.
These commands can be used in Z-probe feature:

[image]

[image]

Super169, 2015-12-02:
Thanks a lot, I will try this one.
As long as there has alternative on it, this request can be ignored.  ^_^

closed 2015-12-02 by Super169

Denvi, 2015-12-02:
> The other way is to save the transformed gcode, then go 0,0,-0.429, then close the application and re-open it. Now the transformed gcode works properly.

You can just go to (0, 0, -0.429) point and then press this one:

[image]

Super169, 2015-12-02:
Thanks a lot.   It works perfectly now.  
That's the best GRBL control software I have ever used.  Thank you very much!