Denvi/Candle #22: No default to imperial

DEF7, 2015-12-12:
As someone who is waiting on a machine to be delivered, I am frightened that I have to trust the screenshots that I will be able to use imperial units instead of metric once it is hooked up to grblControl. The settings->GRBL panel allow the user to change the 'report units' between metric/imperial. But, in the meantime, I cannot play with the SVG files I plan to use (that are huge, and only grblControl can handle them) that I'm importing in imperial units because there is no way to 'trick' the program into switching over from metric. Examining the sourcecode I see how/why it's setup this way, but as someone who is working on a project with tens of thousands of cuts per program, and grblControl being quite literally the *only* software that can handle this without stalling or slowing down, for the machine I am expecting here in the states it would be nice if there was a UI option to change the m_isMetric member of the gcodeparser class to 'true', which only seems to reflect whatever G20/G21 gets sent to the GRBL device. If this could just be forced one way or the other, or maybe even artificially invoke a G20/G21 to grblControl depending on the settings dialog 'report units' option. PS: How can one contact Denvi? Please email me if you get a chance: radioteeth@gmail.com

Denvi, 2015-12-12:
Please, describe a problem in more details. Some explaining images will be very opportunely.

pmpetros, 2016-03-29:
type "g20" into the command line everytime you start up and reset. g20 is the gcode command to read everything in inches. g21 is the same for mm.

closed 2025-09-15 by github-actions[bot]