Denvi/Candle #25: How to install on linux

gregoryperrin, 2015-12-15:
I assumed I had to download and extract the tar.gz linux version, and then execute the grblcontrol file. When i do it the program opens, i select the port, it says "connected" but i can't get the CNC to move (shapeoko 3). Is there some dependencies/librairies i am missing?

I'm on linux mint rafaela 64bits. I've been succesfully been able to use my CNC with other grbl controllers in the last two weeks.

Any clue?

Thank you,

Gregory

Denvi, 2015-12-15:
"Connected" means port is opened so all dependencies in place. 

In normal case you should see "Ready" state after program start.
Looks like it's problem with port name/baud rate.

What baud rate you use? Try to increase "Query period" in settings.

gregoryperrin, 2015-12-15:
I used 115,200. I will to increase "Query period" in settings.

gregoryperrin, 2015-12-15:
I increased, no success. I entered 50, 60, 100, 3000.

Denvi, 2015-12-16:
Post please screenshots:
1) Main window of "grblControl" program after starting.
2) Settings window with "Connection" group visible.
3) Same for other application, which works with your CNC.

gregoryperrin, 2015-12-16:
Hi Denvi, here are the screenshots

main window
----------------
[image]

settings window
--------------------
[image]
Some Universal g-code Sender screenshots
--------------------------------------------------------
[image]
[image]
[image]

Denvi, 2015-12-16:
Good, one more thing. Response to reset command ("Soft reset" - button in UGS).

gregoryperrin, 2015-12-16:
it outputs "Grbl 0.9g"

[image]

Denvi, 2015-12-16:
Custom firmware?
```
// Welcome message
void report_init_message()
{
  printPgmString(PSTR("\r\nGrbl " GRBL_VERSION " ['$' for help]\r\n"));
}
```

Should be "Grbl 0.9g ['$' for help]"

gregoryperrin, 2015-12-16:
No, it´s the Nighlty build, last build available at the moment.

Ok i will try tomorrow, thanks.

Denvi, 2015-12-16:
I thought that "0.9j" version is the latest. "0.9g" - old one.
https://github.com/grbl/grbl

gregoryperrin, 2015-12-16:
I thought you were talking about UGS... 

The GRBL version is the one that shipped with my machine. I didn't upgrade it myself.

I'll try your binary tonight.

gregoryperrin, 2015-12-17:
IT'S WORKING!!!

Nice.

Thanks! What did you change?

Denvi, 2015-12-17:
Just changed sign, detecting GRBL reset procedure.
Also there were problems with "Abort" function. They were fixed in the latest version (0.8.1).
Take care using "M2" & "M30" commands in your G-code programs. They resets GRBL state, because of what the program doesn't display the "Job done" message.

closed 2015-12-24 by Denvi