Denvi/Candle #26: Visualization is not working

vidmo91, 2015-12-19:
Hi,
I have no visualization. I tried check mode, also changed colors.. Nothing helped. Could You please help me?

Denvi, 2015-12-20:
What operation system you using? Post screenshot of non-working visualizer please.

vidmo91, 2015-12-20:
Notebook: Win7 x64, intel HD 4400 graphics, some intel i7 and PC: Win10 x64, radeon HD 3600 graphics, 2x intel xeon 2,5Ghz. Both have same issue.

[image]

Denvi, 2015-12-21:
Looks like your G-code program doesn't have any command which sets "Z" coordinate, i see only "X" & "Y".
Visualizer should know all three coordinates at least once as the "original" point.
So, in your case, you can add something like this to the program start:

```
G0Z0
```

vidmo91, 2015-12-21:
That worked for me. Thank you. I use only XY, because this is laser.

closed 2025-09-15 by github-actions[bot]