Denvi/Candle #27: aborting send program without homing enabled.

DEF7, 2015-12-29:
grblControl fucking rocks, first of all. I just built my X-Carve (waiting for spindle) but I've only been testing it out with my personally modified version of grblControl that runs in imperial, and with a few graphical and keyboard jogging changes.. I will probably not bother using any other GRBL controller program with it because grblControl is *exactly* what I would make if I were to write my own program from scratch (which I was just about to do when I discovered grblControl).

so, I don't have homing switches on my machine, or homing enabled on GRBL, so 'home' is just floating wherever (somewhere outside my house, apparently)

when I run a program and hit the abort button, it seems to perform the Restore XYZ button's function (I assume) and sends the machine off wherever it thinks it should go. What would be great is if the abort button just sent the machine back to wherever it was when it started the program that it was running in the first place, perhaps moving to safety Z first and then rapiding to the original start position.

I'm going to try to hack away at the code and figure this out myself but I figured I may as well contribute my ideas to the project for the project's benefit as a whole.

thanks for sharing grblControl with us, I am working on a project that uses G-code programs with tens of thousands of lines of code, and all the other programs I have tried came to a crawl when I would load my G-code program into them because they were written using tons of external garbage and weren't running natively on my CPU.

please keep up on grblControl! I have been promoting it on various forums telling people it is the best program I have found, because it is true! With the heightmapping feature you're right up there with the best of the other programs.

email me if you want to have some conversations and I'll collaborate because I think grblControl is so extremely close to becoming a 'gold standard' sort of program for GRBL CNCs and I have plenty of professional precision CNC manufacturing experience to contribute, as well as two decades of C programming to help see things through.

radioteeth@gmail.com

Denvi, 2015-12-29:
Many thanks for a good response, glad to try further.

"Abort" function was discussed in issue [#10](https://github.com/Denvi/grblControl/issues/10)
In "two worlds" it's working as follows:

1) "Send" button stores "G92" offsets.
2) "Abort" button sends "Pause" command.
3) Program waits until machine coordinates become invariable.
4) Program moves tools to current machine position to avoid rounding errors.
5) Program restores "G92" offsets.

So, if you have not desirable behavior, look for p. 4.
It's "restoreOffsets()" function in "frmMain.cpp" source file.

DEF7, 2015-12-30:
Ok, I figured out that the problem was with my hacked grblControl that I tried to convert into operating in inches instead of millimeters.

The m_storedX/Y/Z (and probably many other values) were being multiplied by 25.4 by the toMetric() function, which was causing things like RestoreXYZ and Abort to send the machine into space.

Clearly I have failed to make an imperial version of grblControl.

What is the easiest way to make grblControl run in inches? I have discovered all of the G21 commands, and replaced them with G20, and have made a few other little changes, but it seems that metric is woven deep in grblControl.

Thank you!

Denvi, 2015-12-30:
I am going to add imperial units as an option in the next version of the program.
Couldn't help you at this moment, at least because, I don't know how you changed the program.
As far as I remember, toMetric() function used only to handle imperial units from grbl status string.
Offsets stored in metric units, so you should use "G21" command to restore them.
I will deal with this problem more densely in couple of days.

DEF7, 2015-12-30:
I see. I have GRBL running in G20, so everything is being reported in inches, and in the toMetric function it's multiplying everything by 25.4 for the internal conversion back to millimeters so for my version I just had to swap the logic. I'm pretty sure that if m_settings.units was set to metric my version would not work correctly anyway. Thanks!

kquaid, 2016-01-06:
I am having similar problems with my non-homing x-carve and wanted to add to the conversation.  

I really love grblControl, but the Abort button is blocking me at the moment.

When I am testing something and I have to pause, it usually means something has gone wrong.  A bit broke or my settings were bad or what not.  I don't want to ruin my work piece, so I hit "Abort" hoping I can save everything.  Suddenly grblControl tells my x-carve to go very far away.  I really just wish it would abort sending and do nothing else.  I can then use the UI to raise the bit and see what went wrong.

But maybe most people like the current actions when Abort is pressed, so a solution could be to customize the g-code commands in settings.  Most people could leave the default Abort commands in there, and some people like me could just tell the machine to stop and turn off the spindle.  (Plus, doing it this way makes it easier for non-programmers to tweak)

Thank you!  This software is so great!

Denvi, 2016-01-07:
Hello.

>  Suddenly grblControl tells my x-carve to go very far away

Looks like it's a bug. So, please, post:
1) GRBL firmware version.
2) grblControl program version you using.
3) Simple sample G-code program which cause "Abort" button to work wrong.
4) grblControl window screenshots before sending G-code program and after pressing "Abort" button.

pihnat, 2016-01-09:
Hi everyone,
I raised the initial Abort issue in #10 and since it was implemented I've never seen the problem described above - when pressing Abort the CNC machine just stops dead.
Please post the info Denvi requested so the fault can be checked.

kquaid, 2016-01-10:
Very sorry, I wasn't able to get over to my workshop and connect directly to the machine again until today.  Here is what I have:

1) GRBL 0.9i
2) grblControl 0.8.1
3) Attached is a sample NC file I am getting out of VisualCam.  I am using the Mach3 post processor.
4) Attached is a before and after shot of the Console output.  You can see that after hitting Abort it sends 2 more move commands before stopping.

If it is helpful at all, I have noticed that the offset of the last 2 moves is relative to where I pause the program.  It always seems to go the same distance in X Y and Z away from my position at Abort.

[circletest.zip](https://github.com/Denvi/grblControl/files/84161/circletest.zip)
[image]

Thanks!

pihnat, 2016-01-10:
OK it took a while but I can now duplicate the behaviour you describe. It is because of the values in G54 offset coords. I have values of 0,0,0, yours are -8.923,-269.586,22.000. grblControl seems to jump by the offset amount when Abort is pressed. Denvi, is this correct behaviour?

Zero the offsets in G54 by sending  $RST#  to grbl and see how that goes.

kquaid, 2016-01-11:
I am trying everything here and I can not clear G54.  Using $RST# returns "error: Bad number format".  Any reset command from the grbl Wiki does the same, whether I send from grblControl or any serial connection.

It looks like I need to reflash with 0.9j, but then I might not be able to use the x-carve software anymore.

pihnat, 2016-01-11:
Sorry my mistake. The command is  $RST=#
Don't use the other reset commands just this one. The others clear too many settings.
Reference is  https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

kquaid, 2016-01-11:
I tried that one, too.  And $RST=* and $RST=$.  They all gave me the same error and did nothing.

The command has changed from 0.9i to 0.9j, but I can't find anything online that says what the correct command is for 0.9i.

pihnat, 2016-01-11:
I flashed with 0.9i and it did accept the RST command. I'll freshly download it and will try again.

pihnat, 2016-01-11:
With grbl 0.9i, I sent command from grblControl and it works.

[image]

kquaid, 2016-01-11:
Wow, that is just not what I get.  Crazy.

I guess I'll reflash it.  I just hope I get all of the correct settings in there again once it is reflashed.  It seems like some people get new problems when they do this.

Thanks for your help!

pihnat, 2016-01-11:
I used Putty to list my settings and then I did a screen grab and saved as jpg. That way I can always go back to see what they are.
All the best.

kquaid, 2016-01-11:
Good idea.  I'll give it a try.

Thanks!

Denvi, 2016-01-12:
```
G10 L2 X0 Y0 Z0
```

To clear G54 offsets.

kquaid, 2016-01-12:
That fixed it!  All of the offsets are 0, 0, 0 and it stops in place correctly when pressing the Abort button.

So nothing was wrong with grblControl after all.  I look forward to making a lot of cool stuff with this great program.  Thank you all for your help!

Denvi, 2016-01-13:
In fact, G54 offsets as well as others should not affect the functionality of the "Abort" button. Marked as "Bug".

pavelgrigorov, 2016-02-12:
Good day ! a very good program works fine you're done! previously I enjoyed JCNC but downloaded this program and I do not want to work more in the other ! Only I've got a problem , how to set the ratio of pitch to millimeters ? step / mm ??? for example I have the x-axis and y-axis and the z- belt drive stud or different ratio steps that X and Y do please answer would be very grateful ! thanks in advance !!! and if you can throw off please reply to mail pavelgrigorov893@gmail.com

pavelgrigorov, 2016-02-12:
Good day ! Denvi a very good program works fine you're done! previously I enjoyed JCNC but downloaded this program and I do not want to work more in the other ! Only I've got a problem , how to set the ratio of pitch to millimeters ? step / mm ??? for example I have the x-axis and y-axis and the z- belt drive stud or different ratio steps that X and Y do please answer would be very grateful ! thanks in advance !!! and if you can throw off please reply to mail pavelgrigorov893@gmail.com

Denvi, 2016-02-12:
Hi, "step/mm" is a firmware settings.
You can use "$$" command in console to see current GRBL settings:
~~~
$$ < $0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.050 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=250.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=320.000 (z, step/mm)
...
~~~

To change "step/mm" parameter of X-axis, for example, use command:
~~~
$100=400
~~~

pavelgrigorov, 2016-02-12:
Thank you very much! I understood) would not be bad to add this in the
settings to your program has become even more convenient !! you're
cool !!!!

pavelgrigorov, 2016-02-12:
и еще бы неплохо добавить  линейку

2016-02-12 16:34 GMT+03:00 Павел Григоров <pavelgrigorov893@gmail.com>:

Denvi, 2016-02-12:
Создайте новый вопрос (new issue) (вопросы) с подробным описанием предложений. Будет время - займусь. Эта тема немного про другое.

closed 2016-02-19 by Denvi

pavelgrigorov, 2016-02-19:
у меня переводчик плохо работает! что это значит ??

pavelgrigorov, 2016-02-27:
доброго времени суток! я хотел попросить у вас помощи с этим кодом для
ардуино Gy 521 and stepper motor!   я буду очень рад если вы мне поможете

https://www.youtube.com/watch?v=2pbVtUTsgjk

Blocking.pde: In function 'void setup()':

Blocking.pde:75:42: error: 'i2cWrite' was not declared in this scope

Blocking.pde:76:35: error: 'i2cWrite' was not declared in this scope

Blocking.pde:78:34: error: 'i2cRead' was not declared in this scope

Blocking.pde:87:34: error: 'i2cRead' was not declared in this scope

Blocking.pde: In function 'void loop()':

Blocking.pde:133:35: error: 'i2cRead' was not declared in this scope

Ошибка компиляции.
в чем может быть проблема  ???

#include <AccelStepper.h>
#include <Wire.h>
#include "Kalman.h"

#define RESTRICT_PITCH

Kalman kalmanX;
Kalman kalmanY;

///////////stepeer

AccelStepper stepper1(1, 9, 8);

AccelStepper stepper2(1, 7, 6);
int pos = 2000;
//int cantidad = 7;
int addr = 0;

// Define our three input button pins
#define  LEFT_PIN  4
#define  STOP_PIN  3
#define  RIGHT_PIN 5
#define  CONTROL_PIN 11

#define  SPEED_PIN 0

int velocidad = 1000;
int cantidad = 6;
int aceleracion = 500;

/* IMU Data */
double accX, accY, accZ;
double gyroX, gyroY, gyroZ;
int16_t tempRaw;

double gyroXangle, gyroYangle; // Angle calculate using the gyro only
double compAngleX, compAngleY; // Calculated angle using a complementary
filter
double kalAngleX, kalAngleY; // Calculated angle using a Kalman filter

uint32_t timer;
uint8_t i2cData[14]; // Buffer for I2C data

// TODO: Make calibration routine

void setup() {
   pinMode(LEFT_PIN, INPUT_PULLUP);
  pinMode(STOP_PIN, INPUT_PULLUP);
[... 543 lines cut ...]
else

{

if (digitalRead(LEFT_PIN) == 0) {

    stepper2.moveTo(pos);
    stepper2.run();
    }

   else if (digitalRead(RIGHT_PIN) == 0) {

    stepper2.moveTo(-pos);
    stepper2.run();
    }

   else if (digitalRead(STOP_PIN) == 0) {

    stepper2.moveTo(0);
    stepper2.run();
    }

}

}

Denvi, 2016-02-28:
Проблема может быть в отсутствии файлов "i2c.ino" и/или "kalman.h" в папке с приведенным вами .ino файлом.