Denvi/Candle #28: Visualization windows is blank - even text not shown

mariocaptain, 2015-12-30:
Hi,
On my Windows 7 PC the program works fine, visualization works correctly. But on this laptop the window is blank: no text, no coordinate origin, no FPS etc.

Please see attached screenshot.

I thought it has to do with graphic card compatibility, so I have tried changing graphics settings to no avail. On a side note, when I change graphics to 32 bit mode, the application crashs.

I am using a Compaq CQ20 laptop running Windows 7. All drivers are originals from the manufacturer.

Can you please tell me what the problem would be?

Many thanks.
Dave
[image]
[image]

DEF7, 2015-12-30:
My suggestion would be to disable smoothing and MSAA. If the computer is old enough these things can cause problems. Also, try setting the line width to an integer value, like 1, instead of 1.5

Denvi, 2015-12-30:
Hello. First of all try to update (if not yet) your video card drivers with intel one (https://downloadcenter.intel.com/product/81515/Graphics-Drivers-for-Intel-G45-Express-Chipset).

Second (if first doesn't help): download and start "DebugView" program (https://technet.microsoft.com/en-us/sysinternals/debugview.aspx) before you started "grblControl". Post screenshot of "DebugView",
should look like following:

[image]

mariocaptain, 2016-01-04:
Hi Denvi,

Updating the graphics driver with the Intel one solves the problem.
Thanks so much.

Dave

closed 2016-01-13 by Denvi