Denvi/Candle #29: Heightmapping replaces arc moves with linear moves

DEF7, 2016-01-04:
I understand that the G2/G3 commands only work in a single fixed plane, but it would be nice if they could be replaced with multiple linear feeds instead of just flat-out replaced with a single linear feed from start to end. Maybe with the existing code setup this won't be so easy, though, but hopefully I'm wrong. There would probably need to be a new settings panel option for adjusting the length or degree tolerance for converting to lines (just like with the visualizer settings).

Another option is to break up arcs into multiple smaller arcs that lie on the XY plane, and proceed in a step-wise fashion from the Z depth. This would just break one arc into multiple arcs that stair-step according to the interpolated heightmap values, perhaps having a tolerance value for this in the settings panel as well that determines the threshold at which an arc is split, and where it is split. This would be a fantastic feature.

Again, I'm not sure it's entirely possible with the existing setup because it appears that the existing setup is limited to simply modifying the existing g-code commands, instead of re-generating an entire program from scratch that would allow for inserting extra g-code commands (like multiple arcs in place of one arc, for instance).

Let me know what you think. I have experience working with the arc command parameters for g-code editing utilities and might be able to help you figure out how to generate the intermediate arcs that would replace a single arc spanning a varied heightmap area.

Thanks!

Denvi, 2016-01-04:
Hi.

Heightmap modification procedure uses settings from "Parser" group for arc approximation. It replaces all G2/G3 moves with multiple G1 segments. You can set segments size by length or degree.
Give examples if it occurs not so.

I thought about transformation to arc segments, but meanwhile I consider it a low-priority task. Anyway, all arcs will be transformed to lines by GRBL firmware.

DEF7, 2016-01-04:
I apologize, I must have had my degrees too high when I first tried it. I work with very large programs that are made of thousands of arcs and so I turned up the degrees to a higher value to get simpler rendering that is faster, which was causing the heightmapped program to also produce low quality arcs made of very few line segments on the machine. So, it does already work like my first suggestion.

For breaking up arcs into smaller arcs on the XY plane, that only makes sense because the heightmap is in the XY plane, and would be offsetting the smaller arcs along the Z axis to follow the heightmap. For arcs in other planes I think it should just default back to using the line segments the way that it already does..

Overall, I think that the only thing I am concerned about is separating between the length/degrees setting for visualizing arcs, and the setting for what the heightmap uses to generate line segments from arcs. Because for big g-code programs like the ones I am working with I benefit from rendering the arcs very simply with few lines, but also need the heightmapping and the arcs to be round in the machine output, so I am forced to run grblControl slowly rendering many lines unnecessarily. Just separating the settings into two different ones: for the visualizer, and heightmapping, would be great for my own purposes.

Thanks!

Denvi, 2016-01-04:
Well, to divide settings for parser and heightmap is a good idea.

As for visualizer performance at the moment you can use "Simplify geometry" option in settings.

DEF7, 2016-01-04:
Doh! You've thought of everything it seems! I had the Simplify Geometry box checked but the default length value wasn't simplifying the geometry (enough). Fantastic! Now I am completely satisfied with everything. Thanks again!

Denvi, 2016-01-05:
In that case, it is necessary to add new setting like "Streighten max angle" (m_straightenAngle).
And change code to:

```
straight = acos(QVector3D.dotProduct(start.normalized(), next.normalized())) < m_straightenAngle;
```

Angles in radians here, so need to transform to degrees somehow.

iw2fvo, 2017-01-30:
I am very new to this forum.
could I have any detailed procedure for height mapping please ?
Thanks for all
Ambrogio
iw2fvo@yahoo.com