Denvi/Candle #31: GRBLControl dont start on fresh WinXP

Zyrusthc, 2016-02-04:
This is a fresh Windows XP installation with Service Pack 3 installed. Unfortunately GRBLControl not start . What can I do? Athlon XP2400 2GB Ram

[image]

Denvi, 2016-02-05:
1) grblControl version you try to start?
2) What video card & drivers you using?
3) Whether tried 0.5.2 version of grblControl for Windows XP?

Zyrusthc, 2016-02-05:
I have tried all versions the same everywhere .
Radeon 9200SE and last driver from ATI Page

Denvi, 2016-02-05:
I believe grblControl require video card with OpenGL 2.0 support (Radeon 9200SE supports OpenGL 1.4).

But in your case the reason most likely in the processor, which doesn't support SSE2 instructions.
> Athlon XP2400

Zyrusthc, 2016-02-05:
ahh okay thx ,like I have in photoshop also because of the lack sse2 thing .

closed 2016-02-17 by Denvi