Denvi/Candle #32: Mac version ?

Xav83130, 2016-02-04:
Good evening,
it would be possible to compile a version for "Mac"?
If it's not too complicated.
Thank you

Denvi, 2016-02-05:
I think it's possible.
But I haven't Mac, sorry.

Xav83130, 2016-02-05:
you think I could try to do? knowing that I have no knowledge on the subject?

Denvi, 2016-02-05:
Why not?
1) Download Qt for OS X from here: http://download.qt.io/official_releases/qt/5.5/5.5.1/qt-opensource-mac-x64-clang-5.5.1.dmg.mirrorlist
2) Install Qt.
3) Download source code of grblControl.
4) Start Qt Creator and open "grbl_control.pro" project file from "src" directory of sources.
5) Build grblControl (look for image bellow).
6) Post any compiling errors here.

[image]

Xav83130, 2016-02-05:
1 test and 1 error messages
[image]

Denvi, 2016-02-06:
Try to replace "std::isnan" with "qIsNan" in whole project.

Xav83130, 2016-02-06:
after changes

[image]

Denvi, 2016-02-06:
#include <math.h> to top of interpolation.h

Xav83130, 2016-02-06:
result
[image]

Denvi, 2016-02-06:
If replace \<math.h\> with \<cmath\> ?

Xav83130, 2016-02-06:
[image]

Denvi, 2016-02-06:
Sure using \<cmath\>? not \<cmath.h\>.

Xav83130, 2016-02-06:
it's better with <cmath>
[image]

Denvi, 2016-02-06:
Remove
>>> 
#include "math.h"

from "gcodedrawer.cpp" source file. Or replace with #include \<cmath\> if failed to compile.

Xav83130, 2016-02-06:
same result with two
[image]

Denvi, 2016-02-06:
Think need to repeat previous steps for these files:
tooldrawer.cpp
gcodeparser.cpp
gcodepreprocessorutils.cpp

In "heightmapgriddrawer.cpp" line 32 looks the same as:
~~~
if (m_model == NULL || qIsNaN(m_model->data(m_model->index(i, j), Qt::UserRole).toDouble())) {
~~~
?

Xav83130, 2016-02-06:
for "heightmapgriddrawer.cpp" line 50 and 64 also ?

Denvi, 2016-02-06:
Use advanced search:
[image]

Select "Current project", string to find "std::isnan", check "Case sensitive" and press "Find and replace" button:
[image]

Enter "Replace to": "qIsNaN" and press "Replace" button:
[image]

Xav83130, 2016-02-06:
[image]

Denvi, 2016-02-06:
It's easy, just repeat previous step to replace "NAN" with "qQNaN()"

Xav83130, 2016-02-06:
successful compilation

Denvi, 2016-02-06:
How about pressing big green arrow?

Xav83130, 2016-02-06:
the program starts without problems !!!!!
I just connect the milling machine,
it connects.
little problem:
in preferences: colors are all black.
but I can change them manually

Denvi, 2016-02-06:
Need to apply "Default settings" after first start to get normal colors.

Xav83130, 2016-02-06:
it is good with to apply "Default settings".
Now i go test command machine manually.

Xav83130, 2016-02-06:
THE MACHINE MOOOOOOVE is too good

Xav83130, 2016-02-06:
what should he do now?
to install on the Mac.

Denvi, 2016-02-06:
You should use "macdeployqt" util from "QTDIR/bin/" folder to get .dmg file.

Open terminal, change directory to grbl_control build folder with "cd" command (it should be above sources like this one: "build-grbl_control-Desktop_Qt_5_4_2_MinGW_32bit-Release")

~~~
cd path_to_grbl_control.app
~~~

Then:
~~~
$QTDIR/bin/macdeployqt grbl_control.app -dmg
~~~

I beleive it will create .dmg file in build directory.

Xav83130, 2016-02-06:
it is complicated for me ....
I look at all this in one hour

Xav83130, 2016-02-06:
Ok i understand for create a .dmg
but when i run .dmg i have error

"Dyld Error Message:
  Library not loaded: @rpath/QtOpenGL.framework/Versions/5/QtOpenGL
  Referenced from: /Applications/grbl_control.app/Contents/MacOS/grbl_control
  Reason: image not found"

Denvi, 2016-02-06:
May be change "grbl_control.pro" file:
~~~
unix: {
    DEFINES += UNIX #GL_GLEXT_PROTOTYPES
    QMAKE_LFLAGS += "-Wl,-rpath,\'\$$ORIGIN/libs\'"
}
~~~
to 
~~~
unix:!macx {
    DEFINES += UNIX #GL_GLEXT_PROTOTYPES
    QMAKE_LFLAGS += "-Wl,-rpath,\'\$$ORIGIN/libs\'"
}
~~~
will help.

How about just run "grbl_control.app"? Or rebuild, run "macdeployqt" without "-dmg" flag and start "grbl_control.app".

Xav83130, 2016-02-06:
grbl_control.app works.
I still tried to do the .dmg but without success.

Thank you anyway.

Xav83130, 2016-02-08:
When I disconnect the USB computer "grbl control" crash.

Denvi, 2016-11-20:
Fixed in one of the latest commits.

closed 2016-11-20 by Denvi