Denvi/Candle #34: Xubuntu 15.10 run

lavrovSA, 2016-02-14:
I've tried to run grblControl precompiled under Xubuntu 15.10 x64, but unsuccessful - it does not start. Seems that the problem is the stuff asks for i386 Qt libraries, and I don't know how to install all the necessary things....

Regards,
Sergey

Denvi, 2016-02-16:
Output from following command:
~~~
ldd -v grblcontrol
~~~
?

lavrovSA, 2016-02-16:
Here it is.
[lddgrblcontrol.txt.txt](https://github.com/Denvi/grblControl/files/132012/lddgrblcontrol.txt.txt)

At first, there were messages about missing libraries, so I've installed qt5-default:i386 and qttools-dev-tools:i386. After that no complains about libraries, but the message at run - 

>QGtkStyle could not resolve GTK. Make shure you installed the proper libraries.

Then, the GUI with inactive menu and ugly looking appears.

Denvi, 2016-02-16:
Program uses "GTK" visual style on linux, so will look good on Ubuntu for now.
Try start program with "style" key:
~~~
grblcontrol -style fusion
~~~

lavrovSA, 2016-02-16:
Yes, working!

closed 2016-02-17 by Denvi