Denvi/Candle #35: Only rapids are visualized

christoph1703, 2016-02-17:
In version 0.8.1 (Linux), cutting moves are not rendered in the visualization. Running the moves works as expected, only the track is not visible.

[image]

Denvi, 2016-02-17:
Hello, attach G-code program file with the described problem please.

christoph1703, 2016-02-17:
Here is an example (generated with pcb2gcode):
[front.txt](https://github.com/Denvi/grblControl/files/134268/front.txt)

Maybe the terminal output will help:
`locale: "en"`
`shader program created`
`updating geometry 0x9f70748`
`updating geometry 0x9f70870`
`updating geometry 0x9f70748`
`Prepared to load: 1`
`nan point segment added: QVector3D(nan, nan, 2)`
`length nan: 0 0 QVector3D(nan, nan, nan) QVector3D(nan, nan, 2)`
`length nan: 1 1 QVector3D(nan, nan, 2) QVector3D(5.2832, 8.54456, 2)`
`model filled: 18`
`view parser filled: 1`
`updating geometry 0x9f70748`

Denvi, 2016-02-17:
Problem is with commands sequence like:
~~~
G01 Z-0.05000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
F600.00000
X5.28320 Y8.54456
...
~~~

First linear move after "G04" command should contain "G01" prefix for proper work:
~~~
G01 X5.28320 Y8.54456
~~~

Marked as bug. Will be fixed.

christoph1703, 2016-02-17:
Thank you for your quick response! It works if I add the "G01" prefix.

closed 2016-02-19 by Denvi