Denvi/Candle #37: Edge and  hole center finder

milezdr, 2016-03-03:
Hi Denvi you have done awesome work with grblcontrol especially with height map probing.So I wonder if it's possible for you to make edge and hole center finder with this probing routines   http://www.timeguy.com/cradek/01262579508  ?

Denvi, 2016-03-05:
Hello, it's possible. 
But neither grbl firmware nor grblControl app doesn't support RS274/NGC parameters (variables) now.
I don't think that this functionality will be realized soon.
Anyway I will think over the described functions.

milezdr, 2016-03-05:
I talked to the Grbl guy Chamnit what you told me  and he said this : " parameters/variables are easy to simulate with any modern programming language (arguably much better). bCNC does that with their Python-based GUI. If you can program Python, then you can install a plug-in (macro program)."

Denvi, 2016-03-09:
I can do it without Python. 
It is a matter of time and desires.

closed 2016-09-22 by Denvi