Denvi/Candle #39: Numeric keypad 'enter'not recognised

pihnat, 2016-03-05:
Hi Denvi,
When I type a command in the bottom right console area of grblControl's screen I just press the Enter key to send it to the Arduino - no problems.
A colleague said this doesn't work for him and when I checked the difference is that I use the Enter key above the Shift key, he normally uses the Enter key on the Numeric Keypad. If it's the same function ie 'Enter' why doesn't grblControl recognise it?

Denvi, 2016-03-05:
Hello, it is defect. 
Now the program use only "Qt::Key_Return" code to recognize "Enter"-key press, it's necessary to add "Qt::Key_Enter" also.
I will correct it soon.

Denvi, 2016-03-09:
Fixed in commit: https://github.com/Denvi/grblControl/commit/269b49c5c93111239665b5435a4c36417f0bd598

closed 2016-03-15 by pihnat