Denvi/Candle #40: does not work visualization on versions higher than 0.5.2 for xp

Lenrockus, 2016-03-05:
since I had install fresh copy of windows 7 x32,  I'm not able to run correctly grblcontrol (except xp version), visualization doesn't work anymore. what else software should I install to get it work?

Denvi, 2016-03-05:
OpenGL driver.

Denvi, 2016-03-09:
Any motions? 
What graphics card you have installed?

Lenrockus, 2016-03-12:
My graphic card is an old intel 965 express on a compaq presario notebook.it supports only opengl 1.5 and 2.0. When updated to  2.0 grblcontrol crashes on startup. At 1.5 it doesn't show any visualisation

Lenrockus, 2016-03-12:
My graphic card is an old intel 965 express on a compaq presario notebook.it supports only opengl 1.5 and 2.0. When updated to  2.0 grblcontrol crashes on start.

Denvi, 2016-03-12:
"grblControl" works on windows xp well from 0.6. There is no need to compile separate version.
Can you post "DebugView" program output? Refer to #28.

Lenrockus, 2016-03-12:
I will do it as soon as possible, curently I'm not in my home city. Thank you .

closed 2016-04-10 by Denvi