Denvi/Candle #41: 'Connected' status confusing

pihnat, 2016-03-08:
A colleague was setting up grblControl on another computer and we just discovered a small issue.
If the Arduino is NOT plugged into the computer and grblControl is run the first time the default serial port is COM1. Because the computer actually does have COM1 even though nothing is connected to it grblControl responds by displaying 'Connected' in the Status box. Even with the Arduino plugged in if you select a COM port which isn't connected to the Arduino it also displays 'Connected'. When the port is changed to match the Arduino then it finally displays 'Idle'.

This is a little confusing. I can see that the idea is for grblControl to indicate that it is connected to an available serial port but the word 'Connected' implies that it is connected to an Arduino when it really isn't. Would it be possible to display something else rather than the word 'Connected' to make it less confusing?

Denvi, 2016-03-09:
> if you select a COM port which isn't connected to the Arduino

Also with wrong baud rate selected, so program connected to the CNC but can't get status from GRBL firmware.

> NOT to enable the controls unless it's actually communicating with an Arduino

Hard to troubleshoot if only one way communication aviable, only to cnc. Now you can send commands anyway.

> Would it be possible to display something else rather than the word 'Connected'

What is your offer?

pihnat, 2016-03-09:
re your first 2 answers:
yes I understand the problems with one way communication and the limitations of grbl with the current version. I also noticed under some conditions grblControl can display the 'Not connected' message. So is the grblControl start sequence as follows:
- initially it displays 'Not connected'
- it looks for the COM port as set in   settings.ini
- if the port exists it displays 'Connected', enables the controls and sends ? at a rate set in Query Period to get grbl's status
- if it receives a valid reply from grbl it displays the status eg Idle, Alarm, etc

re your answer 3:
Maybe one of these
  CNC not found
  CNC not connected
  grbl not found
  grbl not connected

The alternative is to keep displaying 'Not connected' (meaning it's not connected to grbl) until a valid grbl status is returned.

Denvi, 2016-03-10:
Everything is almost right.

"Not connected"-status displayed:
- Initially
- If program can't open port.

>if the port exists it displays 'Connected', enables the controls and sends ? at a rate set in Query Period to get grbl's status

From one of last versions program doesn't sends next "?" until recieve response from previous.

How about "Port opened"-status? Won't confuse?

pihnat, 2016-03-10:
Yes, that is better because it doesn't imply connected to grbl. So status messages would be:
- "Not connected"
- "Port open" and
- Normal status messages from grbl.

Thanks again
Peter

Denvi, 2016-03-10:
Done in commit: https://github.com/Denvi/grblControl/commit/6cfcd05be56727f1d6bbe10525b100b34091099b

closed 2016-03-15 by pihnat