Denvi/Candle #42: Question to auto leveling option

ojdas, 2016-03-11:
Hi,

I have a question regarding auto leveling set up.
I'm using DIY copper plate where the thickens of plate is 1.5mm.
How to set up auto leveling option to have correct values when the level measurement will be finished?
I should subtract 1.5mm value from the final values after the test?

Thanks,
Piotr

Denvi, 2016-03-12:
The simplest way to mill PCB feature is to use G-code programs wich have origin on top of PCB. So all working moves have negative z-coordinate.

In that case you should:
- set zero z-coordinate on top of copper plate with "Z-probe" function,
- create & apply heightmap,
- mill PCB.

If you have G-code program with an origin in bottom of plate:
- set zero z-coordinate on top of copper plate with "Z-probe" function,
- create & apply heightmap,
- move tool by z-coordinate to 1.0 mm for example (1.0 mm above copper plate),
- correct zero z-coordinate with command

~~~
G92 Z2.5
~~~

where "2.5" is 1.0 mm offset above copper plate + 1.5 mm copper plate thickness

- mill PCB.

milezdr, 2016-03-12:
@Denvi I think you have understand wrong. He is using pcb just for touch off so he can level the surface.

closed 2016-04-10 by Denvi