Denvi/Candle #43: Make with grblControl

Xav83130, 2016-03-13:
No comments.

[image]

[image]

Denvi, 2016-03-13:
Maybe after all comment?

Quality of milling leaves much to be desired. 
I see milling was carried out with insufficient deepening.
Besides, problems with mechanics of the machine are possible.
What router are you using?

Xav83130, 2016-03-13:
indeed,
I have a lot of small problems to settle.
but the circuit is (I hope) functional.

I use a home made machine (with little financial means).
link to my machine: [https://github.com/Xav83130/Mini-CNC](url)

Denvi, 2016-03-13:
Good CNC.
X-Z carriage is weak place. It is necessary to make of metal to get rid of uneven lines when milling.

Xav83130, 2016-03-13:
yes, I also have a little gap in the transmission (despite the backlash)
simple threaded rods.
Finally, with all this I'm pretty happy with the result.
Thank you for grblControl is a good tool.

Denvi, 2016-03-13:
Thanks. It is pleasant to hear that the program was useful.

Xav83130, 2016-03-20:
Hello,
I tried to follow your advice, so I redid the circuit.
it looks better ...
[image]
[image]

closed 2016-04-06 by Xav83130