Denvi/Candle #44: Build problem

pihnat, 2016-03-15:
Hi Denvi,
I've been learning how to use Qt and am modifying grblControl in the process. Everything has been working fine until the last few commits.
I noticed that originally in the source code folder you had many files and 4 subfolders - images, screenshots, shaders and translations. Using this I've had no problems - when I debug in Qt it runs grblControl, connects to the Arduino and i can test everything OK.

However I noticed now with the later versions that you have only some code in your src folder (I assume the main ones which are updated most often) and lots more subfolders. This time everything works OK the first time I run debug. After I close grblControl and press debug again grblControl doesn't connect to the Arduino. It's nearly as if the serial port wasn't closed properly after the last run of debug or maybe the port isn't being open properly. I'm using Qt creator ver 3.5.1.  Any idea what could be causing this?

Denvi, 2016-03-16:
I didn't face such problem.
Any screenshots? First run, second, with console output.

pihnat, 2016-03-18:
OK, with Arduino plugged in and with an empty Debug folder:

I execute Debug, It takes a while to compile but eventually opens grblControl window. grblControl displays "Not connected" in Status. This is displayed in Application Output window of Qt:

[image]

In grblControl I select Service, Settings and click the 'Set to Default' button and change serial port to COM12. grblControl Status goes to "Idle". Here is App Out window now:

[image]

I exit grblControl:

[image]

I now click Debug again. grbl Control window re-opens, Status is "Port opened" ie no comms with Arduino and here's App Out window:

[image]

The only way I can get grblControl to run in Debug mode is to delete contents of Debug folder and re-compile. 
Note that
- this also happens when Release is selected
- this has only started happening after the source folders were changed to the current organisation. Previously this didn't happen.
- when I copy the newly created grbl_control.exe file from Release to replace my old exe file in the folder I normally use to run grblControl I also get "Port opened" Status meaning there's no comms with the Arduino.

Any ideas?

Denvi, 2016-03-18:
Two things i need to know:
1) Try to set wrong port before exiting program, for example "COM1234". Set right port after start. Will it help?
2) Version "0.8.1" sources also have the described problem?

pihnat, 2016-03-18:
1. I have it at work so will test on Monday.
2. Once again I will check on Mon but my impression is that it was OK before the project files were restructured but I'm not 100% sure.

pihnat, 2016-03-20:
I have more details:

**Test conditions**
Arduino is on COM12, COM1 exists but nothing connected to it, latest source 0.8.2.
The following operation is the same in Debug mode, Release mode and also if I copy the released grbl_control.exe file into my existing folder where I normally run grblControl.

**Operation**
If port set to COM12 & status is "Idle", I quit grblControl and re-run then status is "Port opened". The only way to re-connect to the Arduino is to select a different port, press OK then re-select COM12 & press OK. The different port can be COM1 which exists or COM123 which doesn't exist. If COM12 is selected and I re-select it and press OK status is still "Port opened".

So It seems like the only way to connect to the Arduino is to change the port **to** COM12 **from** another port.

I checked previous versions and all was OK up to commits on Feb 19, 2016. The problem starts from commits on Mar 4, 2016:
https://github.com/Denvi/grblControl/tree/9f51945bc62c10c5f08f8adc863fb5ce72719197

Denvi, 2016-03-22:
As you have guessed, it is necessary to revert:

~~~
    // Setup serial port
    connect(&m_serialPort, SIGNAL(error(QSerialPort::SerialPortError)), this, SLOT(onSerialPortError(QSerialPort::SerialPortError)));

    if (m_settings.port() != "") {
        m_serialPort.setPortName(m_settings.port());
    }

    openPort();

    m_serialPort.setBaudRate(m_settings.baud());
    m_serialPort.setParity(QSerialPort::NoParity);
    m_serialPort.setDataBits(QSerialPort::Data8);
    m_serialPort.setFlowControl(QSerialPort::NoFlowControl);
    m_serialPort.setStopBits(QSerialPort::OneStop);
~~~

to:

~~~
     // Setup serial port
     m_serialPort.setParity(QSerialPort::NoParity);
     m_serialPort.setDataBits(QSerialPort::Data8);
     m_serialPort.setFlowControl(QSerialPort::NoFlowControl);
     m_serialPort.setStopBits(QSerialPort::OneStop);
-    if (m_settings.port() != "") {
-        m_serialPort.setPortName(m_settings.port());
-        m_serialPort.setBaudRate(m_settings.baud());
-    }

   connect(&m_serialPort, SIGNAL(error(QSerialPort::SerialPortError)), this, SLOT(onSerialPortError(QSerialPort::SerialPortError)));
~~~

pihnat, 2016-03-22:
In frmmain.cpp (0.8.2)  I commented out lines 136 to 151 (I removed line 151 too which is "connect(&m_serialPort ..... ) and inserted the new lines but now it doesn't connect to Arduino at all.
Shouldn't there be an "openPort()" command somewhere in the replacement commands?

Denvi, 2016-03-23:
> Shouldn't there be an "openPort()" command somewhere in the replacement commands

No.
Line 151 should stay.

Look for last commit.

pihnat, 2016-03-24:
OK the commit shows it clearly and I see what I typed incorrectly. It works perfectly now.
Thanks again for your prompt reply.
regards
Peter

closed 2016-03-24 by pihnat