Denvi/Candle #47: GRBL stuck during work

ojdas, 2016-04-04:
Hi,

I have an issue when I'm using this tool during cutting ply.
I have observed very strange situation - grbl stuck during ordinary works.
It is very annoying when the whole job we need do once again :(

Do you have any idea why?

thx.
Piotr

Denvi, 2016-04-05:
Hello.

- What CNC you using?
- How it's stucks? Steppers just stopes? Spindle stops? Any messages in console panel of grblControl? Post screenshot of main window when stuck.
- Try to do idle running on same G-code program with spindle turned off.
- CNC have end-stops? Try idle running with disconnected end-stops.

ojdas, 2016-04-05:
Hi Denvi,

I'm using Arduino CNC shield V3
Steppers just stops, spindle is not connected to shield and power supply for spindle is separate.
I have end-stops so I will try to disconnect during running but it is quite strange because during running all end-stops are not touched.
Previously I have used GRBLControl v8.1 today I will try 8.3 maybe here is an issue?

Thanks,
Piotr

Denvi, 2016-04-05:
> but it is quite strange because during running all end-stops are not touched.

There are suspicions that has put in hindrances on wires.

> Previously I have used GRBLControl v8.1 today I will try 8.3 maybe here is an issue?

I don't think there is any differences in that versions releative to your problem.

closed 2016-05-25 by Denvi