Denvi/Candle #48: Gcode generated with J tech photonics wont show

electrophile, 2016-04-10:
This is a rather quirky issue. I'm using grblControl to control my home built laser engraver. When I create gcode with the extension that comes built in with Inkscape, everything works great (except I cannot automate the multiple passes). When I create gcode using J Tech photonics inkscape extension (which can automate multiple passes), and load it in grblControl, I cannot see any of the lines that the engraver is supposed to engrave (unlike the one with the one built in with Inkscape which shows me all the paths). grblControl in this case does finally give the needed output, it just does not show the paths on screen. Any ideas?

Denvi, 2016-04-10:
Hello, may be #26 will help you.

electrophile, 2016-04-10:
That helped! Thanks :+1:

closed 2016-04-10 by electrophile