Denvi/Candle #49: Build and Debugger

ser007, 2016-04-11:
Denvi, congratulations for your project! Loved it! Its interface is very clean and the controls are simple.

I make one fork of your project to study and try to implement some changes, but after reinstalling my Qt Creator 4 times I am unable to compile your code.

I'm using Windows 7 with MinGW, the Qt5.6.0 and Qt Creator 3.6.1.

Please. What is the best way to build and debugger?

Thank you.

Denvi, 2016-04-11:
Hello, thanks for your responce.

I'm using Qt 5.4.2 with MinGW 4.9.1 compiler, GNU gdb 7.8 debugger.

>  I am unable to compile your code.

What errors do you have while compiling?

ser007, 2016-04-11:
Denvi, 

I realized download the Qt 5.4.2 with MinGW 4.9.1 compiler with you mentioned through the link 
[qt-opensource-windows-x86-mingw491_opengl-5.4.2.exe
](https://download.qt.io/official_releases/qt/5.4/5.4.2/qt-opensource-windows-x86-mingw491_opengl-5.4.2.exe.mirrorlist)

Performed the installation and there was no problem. Now build and debugger! :+1: 

I will begin studies.

Thanks for your help.

closed 2016-09-11 by Denvi