Denvi/Candle #50: cutting speed

zagranichnyy, 2016-04-19:
you can explain to me as redefinition of speed works? I am admissible I consider that it has to be equal 100% and already a hedgehog from the maximum speed it would be possible to reduce, and now I can't understand as it works

Denvi, 2016-04-21:
Let's say you have the following G-code program:

~~~
G21G90
G0X0Y0Z0
G1Z-1F30
G1X10F300
G0Z1
~~~

Here we have 2 feed settings: 30 mm/min for z-axis movement and 300 for x-movement.

With feed override disabled or set to 100% these feed will be equal to 30 & 300 respectively.

With feed override 200% feeds will be 60 & 600.
50% will give 15 & 150.

zagranichnyy, 2016-04-21:
It is clear, but why value of 500%, I think that 120-150% would be enough

milezdr, 2016-04-21:
I think is good (0-500%) because you can have very wide selection of speed when you are beginner or using new materials on your cnc .

zagranichnyy, 2016-04-21:
Well on taste and color felt-tip pens different. And whether I can at myself somehow edit this parameter that 100% were displayed

Denvi, 2016-04-21:
> edit this parameter that 100% were displayed

I don't understand your question. Describe in more detail please.

zagranichnyy, 2016-04-23:
можно написать по русски?

Denvi, 2016-04-23:
Можно. Я не понял часть вопроса про отображение 100%.

zagranichnyy, 2016-04-24:
Я имел ввиду смогу ли я как нибудь отредактировать вашу программу, чтоб отображалась макс скорость 100%

Denvi, 2016-04-24:
Вам нужно будет скомпилировать исполняемый файл "под себя", предварительно скачав исходный код и отредактировав форму главного окна "frmmain.ui" в дизайнере форм.
Ниже приведено изображение с полем, в котором необходимо будет заменить максимальное значение с 500 на 100. Под полем имеется ползунок, у него также нужно заменить максимальное значение.

[image]

zagranichnyy, 2016-04-24:
А какое нужно для этого программное обеспечение?

zagranichnyy, 2016-04-25:
Я установил qt но он выдает ошибку компиляции

Denvi, 2016-04-25:
Посмотрите #49.
Если не поможет, предоставьте снимок экрана с ошибками компиляции. Будем разбираться.

closed 2016-05-02 by zagranichnyy