Denvi/Candle #52: Restore X/Y/Z sends Z in the wrong direction

corradovt, 2016-05-11:
ref: https://github.com/Denvi/grblControl/issues/10 - doesn't seem to apply to my condition.

Caveat: I am a brand new n00b to CNCs and grblControl. I have an extremely low-end PCB CNC. The Arduino is running GRLB 0.9J.

I've figured out how to edit the Stored X/Y/Z in the INI file and have synced that up to be the left-most, front-most, bottom-most corner of the cutter's position. Said another way, the spindle can only be moved up, right, and back from it's current position.

When I enter the program, I confirm both my Machine Coordinates are 0/0/0 and the mouse-hover over Restore X/Y/Z are 0,0,0.
- When I move the spindle in either the X or Y (or both) directions and click, "Restore XYZ", the CNC comes perfectly back to 0/0/0. 
- When I move the spindle along the Z axis, whether it be alone or with X and/or Y, and click "Restore XYZ" the Z-axis takes off in the wrong direction and until it maxes out at the top (currently limit switches are not available).

I've also noticed that Aborting a job sends the spindle off in the extreme opposite direction from 0/0/0 to a point that appears to far exceed that range of my very-small machine (4" x 7" x 2" travels). Can this be configured anywhere to recognize the $130 - $132 "max X/Y/Z" parameters as the "Safe" position?

Denvi, 2016-05-14:
Hello, post please some screenshots (before & during issue) and console output (checkboxes in "Console" group of settings must be set) for:
> When I move the spindle along the Z axis, whether it be alone or with X and/or Y, and click "Restore XYZ" the Z-axis takes off in the wrong direction and until it maxes out at the top (currently limit switches are not available).

corradovt, 2016-05-15:
Thanks for the reply, Devi! I realized I was a few point releases behind so I retested tonight with v0.8.3.

My mic wasn't being cooperative tonight, but I uploaded videos of my workflow here for each scenario here.

https://drive.google.com/folderview?id=0ByxA1mytYa_RU0N4M3huSEVqUHM&usp=sharing

In video 1, I start with my CNC physically at 0,0,0, matching the program. When I move X & Y 2.000 in the positive direction and Restore XYZ, the spindle returns to 0,0,0; when I move Z 2.000 in the positive direction, and Restore XYZ, it flies off in the positive direction of travel instead of returning to 0,0,0 like I expected.

In video 2, I load a piece, start job and attempt to notate with the mouse the "limits" I've tried to set for the MAX X, Y, Z for the CNC. My expectation is that when I Abort the job, the spindle will travel to the MAX X, Y, Z as defined or at least respect the limits defined in the grbl configuration. As I show in the video, when I PAUSE and ABORT, the spindle travels to a range beyond the realm defined for the machine.

I'm not 100% certain my board has functional limit switch capability (it has the ports, but no apparent routes on the circuit board to support them), so assume that I don't have those safeguards available.

This the output from my CNC for settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=150.000 (x max travel, mm)
$131=90.000 (y max travel, mm)
$132=30.000 (z max travel, mm)
ok

Xav83130, 2016-05-15:
hello,
the solution may be the next in the file "config.h" of "grbl".

Denvi, 2016-05-15:
Ok, video is good, but i didn't see any console output on button presses.

So, what I have with
> when I move Z 2.000 in the positive direction, and Restore XYZ, it flies off in the positive direction of travel instead of returning to 0,0,0

[image]

"Console" settings:

[image]

corradovt, 2016-05-15:
Gotcha, I didn't understand what you were asking for with the Console. My apologies.

My notations added with //
[CTRL+X] < Grbl 0.9j ['$' for help] // initial load
G21 < ok // Click "Restore XYZ"
G53G90G0X0Y0Z0 < ok // Machine Executes Restore XYZ (no action; machine was already at Zero)
G92X0Y0Z0 < ok
G21 < ok // Click "Safe Z" (set at 10.000)
G53G90G0Z10 < ok // Machine Executes Safe Z (functions as expected)
G21 < ok // Click "Restore XYZ" -- expect same result as previous Restore XYZ
G53G90G0X0Y0Z254 < ok // Machine attempts to navigate to Z value of 254.000; also updates Work Coordinates to Z= 254.000
G92X0Y0Z254 < ok 

When I run with without the motors activates, when I click Restore XYZ again, this is resultant output to console:
G53G90G0X0Y0Z6451.6 < ok
G92X0Y0Z6451.6 < ok

-------

For the "Abort" here's what I see. From a Paused job and a clear console, this is the entire output when I click "Abort."

[CTRL+X] < Grbl 0.9j ['$' for help]
G21G53G90X0.2794Y0.0762Z792.632 < ok
G21G92X0.2794Y0.0762Z792.632 < ok
G54 G17 G21 G90 G94 M9 T0 < ok
S3000 < ok.

Thank you again for you quick reply. I've also attached my Settings.INI file (added. (.)TXT to pass the file-format check).
[settings.ini.txt](https://github.com/Denvi/grblControl/files/265087/settings.ini.txt)

Let me know if you need anything more.

Denvi, 2016-05-15:
Set "Report units" to "Metric" in settings.

Now you have metric units in grbl firmware ($$-command output):
>$13=0 (report inches, bool)

and inches in grblControl (settings.ini):
> units=1

[image]

corradovt, 2016-05-15:
So realized in the shower that the 254.000 was an Imperial to Metric conversion. However, when I change everything to Metric (which is NOT desired since I work in Imperial Units), the Restore XYZ does nothing. 

Starting from 0,0,10 and clicking Restore XYZ, the spindle doesn't move and returns this to the console.

G21 < ok
G53G90G0X0Y0Z10 < ok
G92X0Y0Z10 < ok

Denvi, 2016-05-16:
"Restore XYZ" restores G92 offsets (which has been set with buttons "Zero XY" & "Zero Z") and moves spindle in XY-plane to (0,0).

corradovt, 2016-05-16:
Would it be more accurate to refer to the function as "Restore XY?" 

Is there a User Guide or "Help File" posted anywhere that describes the features & functions of the program and the setting/configuration options? Trial & error and guesswork is becoming a tedious method for learning the program.

Denvi, 2016-05-16:
There is no User Guide for now. In plans.

closed 2016-09-11 by Denvi