Denvi/Candle #53: Program crash when Carvey door opens

flety-emmanuel, 2016-05-18:
Greetings, I love the program, finally got GRBL clients to work with my Carvey after changing the report line. Carvey uses gCarving fork (GRBL 1.0 based), I only had to change the message the client expects to read the GRBL version and know that it has "booted".
I can do many things now with GRBL control, however when I open the carvey door / front lid, it outputs a message [Check Door] and Grbl control crashes.
A fix would be awesome

Denvi, 2016-05-19:
Hi, fixed in [0.8.4](https://github.com/Denvi/grblControl/releases/tag/v0.8.4) version release.

flety-emmanuel, 2016-05-20:
thank you so much. Side question : do you think you'll support other HW based on GRBL but who don't reply with the GRBL "boot string" ? Cavey is one of those things (that what we have). I've flashed it so that GRBL boots with a standard string (version) but now of course, Carvey's base software (easel) doesn't recognizes it. I know it's chicken and egg question but i figure I'd ask. Thanks again for the prompt answer !

Denvi, 2016-05-20:
I have added "gCarvin" init string. Test the attached file please.

[grblControl_gCarvin.zip](https://github.com/Denvi/grblControl/files/274643/grblControl_gCarvin.zip)

closed 2016-09-11 by Denvi