Denvi/Candle #54: Language Pref

patgadget, 2016-05-19:
Hi,
thanks for the Wonderful Program.
i would like to create some print screen of the program and add information in you wiki page.
but my computer is in french and your program adopt it and i cannot find a way to change the language to English.

Is there a way to change the language on your software?
Thanks a lot again for grblControl.

Pat Gadget.

Denvi, 2016-05-19:
Hello, application has english language by default.

So easiest way - to rename "translations" folder.

[image]

patgadget, 2016-05-24:
Perfect, i can figure out how to insert a print screen,
is there and easy way? or could you send me to a explanatory page?
Hope my Wiki Pages Helps?
Thanks

Denvi, 2016-05-24:
I think we need store screenshots on github, you can clone repo & add all needed screenshots to "wiki/images" folder through the pull request.

closed 2016-09-11 by Denvi