Denvi/Candle #55: I can not see the graphic in the window

jvarl, 2016-05-19:
Hello, Could you please help me?
I installed grblcontrol on two PCs, one with windows 7 and other Windows 10. In the first, the application works well. In the second pc, the program works well only on the part of commands and file upload, but I can not see anything in the drawing window (aside from the five icons). Re-install the video driver recommended by intel 64 bits, with the same result. What could I do next?

Denvi, 2016-05-20:
Hi, what specs of the second pc (graphics card)?

Refer to #28.

jvarl, 2016-05-20:
Second PC  with windows 10 is  HP PRO:  Intel Pentium II dual core cpu, Intel® Virtualization Tech G45/G43 express chipset (corporation-WDDM 1.1)
I installed also grblcontrol in two other laptops, with the following results:
3th PC Laptop COMMODORE with Windows 10: Intel Pentium dual T3200 @ 2.00 GHz cpu. Phoenix TrustedCore (Toshiba) with Microsoft basic video adapter. Result:  Same the second PC, no show graphic image.
4th PC Laptop HP Pavilon with windows 10 home.  Intel core duo P7550 @2.26 GHz cpu. Video adapter ATI MobilitY Radeon HD4650. Result:  Grblcontrol runs very well !! same as the first PC with windows 7. 
I don’t know what is happened with installings in 2th and 3th PC cases. Thanks a lot for your answer.

Denvi, 2016-05-20:
grblControl needs OpenGL 2.0 support, so you should have video adapter with OpenGL 2.0 and appropriate drivers.

jvarl, 2016-05-20:
Oh, now I can understand the problem! Graphic video support to run grblControl must overcome OpenGL 2.0!  So now I installed GLview (OpenGL Extensions Wiewer 4.4) program in the four PCs mentioned and tried the respective versions compliant. Results: first PC version 3.1, second PC: version 1.1, third PC: version 1.1, fourth PC: Version 3.3. The only solution is to upgrade the hardware on 2nd and 3rd PCs. I think, grblControl is a very good program. Thank you very much for your respectful attention.

closed 2016-05-28 by jvarl