Denvi/Candle #57: Need build instructions

baruch, 2016-05-25:
I couldn't locate build instructions and failed to build the program.

The pre-built code doesn't work for me on my 64bit Linux and qmake (which I just guessed is used) complains about serialport and I couldn't see an easy way to use/install it. Maybe it can be put as part of the repository?

Denvi, 2016-05-26:
> complains about serialport 

What exactly error messages you have?

Hello, you can refer to #32. Of cause need to download appropriate Qt version for linux.

baruch, 2016-05-27:
I get the message `Project MESSAGE: Warning: unknown QT: serialport` from qmake and then build fails with:

```
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -DUNIX -DsNan="65536" -DAPP_VERSION=\"0.8.4\" -D_USE_MATH_DEFINES -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/qt4 -I/usr/X11R6/include -I. -I. -o main.o main.cpp
In file included from main.cpp:18:0:
frmmain.h:8:36: fatal error: QtSerialPort/QSerialPort: No such file or directory
compilation terminated.
```

baruch, 2016-05-27:
None of the issues in #32 seem to be relevant to this issue.

Denvi, 2016-05-27:
What Qt framework version you have installed?

baruch, 2016-05-27:
I'm using debian unstable and have both qt4 and qt5 installed. qmake is from qt4 (4.8.7). The makefile mentions qt4 libraries and includes.

baruch, 2016-05-27:
I actively switched to QT5 with `export QT_SELECT=5` and now it works. I need to do `qmake && make clean && make -j4` do get the build working to remove the remnants of the qt4 build attempts.

It would be nice if you documented this in the readme to save time to others.

Denvi, 2016-05-27:
Ok, will be added.

closed 2016-05-27 by baruch