Denvi/Candle #58: Как настроить задержку старта шпинделя

zagranichnyy, 2016-06-10:
Как сделать чтоб программа при отправки задачи включила шпиндель и после 15 секунд начала выполнять задание.

Denvi, 2016-06-10:
Hi, you can use G4 - command to add delay. 
For example:

~~~
M3S10000
G4P15
...
~~~

zagranichnyy, 2016-06-10:
А как сделать чтоб это было каждый раз в автоматическом режиме?

Denvi, 2016-06-10:
Это уже вопрос к программе, в которой подготавливаются g-code файлы.

zagranichnyy, 2016-06-10:
Использую vectric aspire, не замечал вроде настройки задержки. Завтра еще раз пересмотрю. Но чувствую что тогда постпроцессор придется редактировать.
Еще вопрос, а Вы не собираетесь менять внешний вид программы? Некоторые элементы не удобны. Но все равно мне нравятся Ваши труды! Спасибо!

Denvi, 2016-06-10:
Пока никто вроде не жаловался на интерфейс.
Какие элементы не удобны?

zagranichnyy, 2016-06-13:
помогите пожалуйста с правильной постановкой кода задержки, у меня выходит ожидание 15 сек и потом запускается шпиндель, а мне нужно чтоб 15 секунд было после запуска шпинделя, ниже прикреплен постпроцессор,

`+================================================
+                                                
+ G Code - Vectric machine output configuration file   
+                                                
+================================================
+                                                
+ History                                        
+                                                
+ Who    When       What                         
+ ======== ========== ===========================
+ Tony     02/08/2005 Written                      
+================================================

POST_NAME = "G Code (mm) (*.tap)"

FILE_EXTENSION = "tap"

UNITS = "MM"

+------------------------------------------------
+    Line terminating characters                 
+------------------------------------------------

LINE_ENDING = "[13][10]"

+------------------------------------------------
+    Block numbering                             
+------------------------------------------------

LINE_NUMBER_START     = 0
LINE_NUMBER_INCREMENT = 10
LINE_NUMBER_MAXIMUM = 999999

+================================================
+                                                
+    Formating for variables                     
+                                                
+================================================

VAR LINE_NUMBER = [N|A|N|1.0]
VAR SPINDLE_SPEED = [S|A|S|1.0]
VAR FEED_RATE = [F|C|F|1.1]
VAR X_POSITION = [X|C|X|1.3]
VAR Y_POSITION = [Y|C|Y|1.3]
VAR Z_POSITION = [Z|C|Z|1.3]
VAR X_HOME_POSITION = [XH|A|X|1.3]
VAR Y_HOME_POSITION = [YH|A|Y|1.3]
VAR Z_HOME_POSITION = [ZH|A|Z|1.3]

+================================================
+                                                
+    Block definitions for toolpath output       
+                                                
+================================================

+---------------------------------------------------
+  Commands output at the start of the file
+---------------------------------------------------

begin HEADER

"T1M6"
"G17"
"G0[ZH]"
"G0[XH][YH][S]M3"

+---------------------------------------------------
+  Commands output for rapid moves 
+---------------------------------------------------

begin RAPID_MOVE

"G0[X][Y][Z]"

+---------------------------------------------------
+  Commands output for the first feed rate move
+---------------------------------------------------

begin FIRST_FEED_MOVE

"G1[X][Y][Z][F]"

+---------------------------------------------------
+  Commands output for feed rate moves
+---------------------------------------------------

begin FEED_MOVE

"[X][Y][Z]"

+---------------------------------------------------
+  Commands output at the end of the file
+---------------------------------------------------

begin FOOTER

"G0[ZH]"
"G0[XH][YH]"
"M30"

`

Denvi, 2016-06-16:
Что-то я в вашем постпроцессоре не наблюдаю команды G4, откуда у вас задержка в 15 секунд берется?

>"T1M6"
"G17"
"G0[ZH]"
"G0[XH][YH][S]M3"
>

Думаю, после этих команд нужно добавить

~~~
G4P15
~~~

С Vectric не работал, проверить не на чем.

zagranichnyy, 2016-06-16:
Я нашел другой выход. Просто уменьшил в настройках инвертора время разгона до 3 секунд. Теперь он успевает разогнаться до входа в материал.

closed 2016-06-17 by zagranichnyy