Denvi/Candle #60: Hard Reset/Abort

Robyrob, 2016-06-10:
Dear Denvi, 
I noticed a different behavior when I press the reset button of the program and when I hit the reset button on the Arduino Nano pin A0
In particular, reset from the program works perfectly, but I just get a message "error: Alarm Lock" in the console and the status doesn't changes using the hardware Reset/Abort key (A0)
Similar behavior using "Hard limit"
Thank you

Denvi, 2016-06-10:
Hello, if have fixed status updates on hardware reset in last commit.
But "Hard limit" state of GRBL blocks any subsequent requests of the status until hardware/software reset. So i can't get any status except previous on hard limits.

Robyrob, 2016-06-10:
Thank you for solving status update on hardware reset. I just tried it and now works ok.
About limit, I understand GRBL blocks status request, that's pity.
I hope you can find a solution to update the status to "Alarm" or "Limit error". I think will be helpful to quick understand something was wrong

closed 2016-09-11 by Denvi