Denvi/Candle #61: grbl control won't start on lubuntu

leonardocecchin, 2016-06-10:
Hi, I cannot open the software on lubuntu latest version, here is a photo of the terminal
Thanks, leonardo

leonardocecchin, 2016-06-10:
[image]

leonardocecchin, 2016-06-11:
[image]
Here is it, I started both with superuser or not, the error remains the same, also with older versions of the software

Denvi, 2016-06-11:
Please post output from the following command:

~~~
ldd -v grblcontrol
~~~

And what hardware you use (cpu, graphics card)?

leonardocecchin, 2016-06-11:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
 These are the screenshots of the output, the pc is an old Intel Pentium that natively was running windows XP, anthe graphics card is the original one, there isn't a code or name or something
 Thanks

Denvi, 2016-06-16:
Install Glxinfo and check supported OpenGL version please.

~~~
sudo apt-get install mesa-utils
glxinfo | grep "OpenGL version"
~~~

CPU info also needed:

~~~
cat /proc/cpuinfo
~~~

jflapao, 2016-07-16:
Hello is possible your software work on a raspberry pi the model B with 512mb or any other more powerfull

Denvi, 2016-07-30:
Not for now. All operation in this direction is frozen.

closed 2016-09-11 by Denvi