Denvi/Candle #62: UI Suggestions

intensite, 2016-07-15:
Hello,

First congrats and thank you for that fine piece of software.  Just discovered it today.

If you permit, I would like to suggest a few improvements to the UI.

1. On the state panel, the work and machine coordinate boxes could be resized slightly smaller to allow a label for the axis they represent  i.e( X: 140  Y: 39  Z: 10)  

2. On Windows 10, on the right hand side panel (Jog, Heightmap, Console), the scrollbar is way too thin (1 or 2 pixels).  It is very difficult to reach.  Since my laptop has a relatively small resolution, I unfortunately need to scroll this section often.  Same problem in the settings page.

3. Related to item 2, could the console be moved in its own TAB with the GCode commands underneath the visualisation.  This way, it would increase the console's size significantly while giving more space for the panel widgets (Jog, Heightmap, Feed & Spindle).  I find much more useful being able to view and send manual commands in the console than seeing the GCode of a work file anyway.

4. Safe Z button.  It should be mentioned (maybe in settings screen) that machine coordinate are used for this function.  At first, I thought it was in relation to the work coordinates. In my case, Z works in negative so the default value of 10 was hitting my upper limit switch with full force.  I changed it to -10 and it now stops 10mm before hitting the limit.

5. Have you considered a "macro" function where one could store regularly used G-Code commands.

6. I have noticed that in the Jog section, the checkbox for keyboard control doesn't seem to work (at least not on a Windows laptop arrow keys).

7. Finally a small tooltip on the settings screen items could be useful, specially on the parser, tool model and visualisation sections.

None of these are show stoppers, but they would make the this great application almost perfect.

I am a developer myself but I have zero experience with QT.  Good time to learn it I guess :-)

Keep up the good work.

Stephen..

Denvi, 2016-07-30:
Big thanks for your notes.

> the scrollbar is way too thin 

You can drag panels without scrollbar, by mouse wheel, or by dragging panel contents. The same for settings form.

> I have noticed that in the Jog section, the checkbox for keyboard control doesn't seem to work

Bindings for numpad:

[image]

jflapao, 2016-08-24:
Hello, and great job, I'm intent to use the software on a small netbook 10" screen, but the screen is to small the work and machine coordinates could be more small is only to indicate not to use, the command buttons could be more small too, like "intensite" say the console be moved in its own TAB with the GCode commands underneath the visualisation, and like that maybe we have room to the control pad. The macro to store g-codes is a good idea, but   a memory to store the machine positions is a good one to, imagine i make a job with two or more different tools is a good idea store the start job coordinates to start again with the new tool.

Denvi, 2016-08-25:
Hi.

Denvi, 2016-09-22:
TODO:
- Rework main window layout for wide screen monitors
- Move console window to own tab/frame
- Add "Positions" panel
- Realize enhanced g-code command processing with variables support
- Add "Macro/subprograms" panel

ASHIT-AXAR, 2020-01-15:
do we need some extra control buttons ? :thinking: :thinking: 

a find/goto job XY center button (probing at corner is not the suitable option always)
or a mini popup window to quick select/goto a corner (X,+, center)
a goto work_zero button
a park button (machine goes to a predefined coordinate before shutdown)

closed 2025-09-15 by github-actions[bot]