Denvi/Candle #63: Manual

fgeer, 2016-08-07:
Hello, 

I'm new to gbrlcontrol, is there a manual about this program and how to work with the grafical interface in combination with my cnc. I have no clue what all the knobs are. I only want to know how the interface works, not the gbrl commands.

Denvi, 2016-08-14:
Hi, I will be engaged in a manual (Wiki) as soon as possible.

Peter6891, 2016-11-19:
yes, the manual would not hurt ... it is difficult to deal with the program button functions)

pswarman, 2017-03-24:
Sorry Denvi, but the guys are right, I started looking at other products because I couldn't find a probing function in your program; it wasn't obvious that the program could do that. I only found out after a week of trying other software packages. A manual will win you more users. I should also say, the more I get to use your program, the more I like it. Sincere thanks. Paul (cnc 2418 /Arduino user)