Denvi/Candle #64: Visualization is not working

tacitus-def, 2016-08-14:
Hi,
I have no visualization after opening files. Only white space without any cursor and labels.
[image]

OS: Windows 7 64bit Pro
Videocard: Mobile Intel(R) 965 Express Chipset Family (OpenGL 2.0 Supported)

Denvi, 2016-08-14:
Installation of latest Intel drivers should solve problem:
https://downloadcenter.intel.com/download/18228/Intel-Graphics-Media-Accelerator-Driver-for-Windows-7-64-exe-?product=81508

tacitus-def, 2016-08-14:
This latest driver for videocard is already installed on my laptop.

Denvi, 2016-08-14:
How about "GPU Caps Viewer" output?

[image]

misisl, 2016-08-14:
same problem...
W7 64bit, APU Radeon HD 8570D. Newest AMD driver "amd-catalyst-15.7.1-with-dotnet45-win7-64bit". 

[image]

Denvi, 2016-08-14:
Ok, need "DebugView" output also #28.

misisl, 2016-08-14:
[image]
[grblCntrl.zip](https://github.com/Denvi/grblControl/files/417145/grblCntrl.zip)

Denvi, 2016-08-14:
> line 6 and other after opening gcode...

I cant see any problem with Open GL for your case. Have you "tool represenation" & axes on your visualizer window? Looks like #26 relative problem.

misisl, 2016-08-14:
Yes, same as #26.  Sorry, i am idiot.

tacitus-def, 2016-08-15:
DebugView
[image]
GPU Caps Viewer
[image]

acaballero6270, 2016-08-17:
Make sure your G code sets Z. Example: G0X0Y0Z0. See FAQ section in wiki

tacitus-def, 2016-08-17:
I see only white space without any cursor and labels, which should seem even in an empty file. On another computer with the same settings and the same file works fine.

Denvi, 2016-08-17:
I will test program on 965 chipset within two days.

acaballero6270, 2016-08-17:
Maybe you could try this as a workaround. In my computer (Windows 7) if I right click on grblControl.exe a menu pops up and there is an option that says "run with graphics processor". If your computer has more than one you may choose  another.

acaballero6270, 2016-08-17:
FYI, if I choose the intel integrated graphics I get the same problem. If I leave the default (NVIDIA) it works fine.

tacitus-def, 2016-08-18:
No, laptop have only one integrated videocard.

Denvi, 2016-08-19:
I found the suitable PC (not my one) with a similar problem. Looks like it is glsl version problem. I will try to deal with it next week.

Denvi, 2016-09-02:
@tacitus-def 
Check attached file please.

[grblControl#64.zip](https://github.com/Denvi/grblControl/files/452726/grblControl.64.zip)

acaballero6270, 2016-09-03:
I tried in my computer. Still the same problem with the Intel graphics

Denvi, 2016-09-03:
Next try (the problem is that I do not have Intel 965).

[grblControl#64_1.zip](https://github.com/Denvi/grblControl/files/453459/grblControl.64_1.zip)

Denvi, 2016-09-03:
Next try. Recompiled with dynamic OpenGL/Angle/Software rendering.

[grblControl#64_1.zip](https://github.com/Denvi/grblControl/files/453527/grblControl.64_1.zip)

acaballero6270, 2016-09-03:
Congratulations. It works!

BuiChien, 2016-09-22:
Dear Denvi,

Could you help me how can I recompile your source with dynamic  OpenGL/Angle/Software rendering.

Denvi, 2016-09-22:
- Clone repo
- Checkout "iss64" branch
- Build with Qt 5.7.0 (MinGW 5.3.0)

BuiChien, 2016-10-06:
Thanks @Denvi

tacitus-def, 2016-10-07:
Thanks @Denvi, it works!

closed 2016-10-07 by tacitus-def

Dante-666, 2017-11-12:
I am getting a similar issue on a Linux machine, I tried #26, I also tried checking out iss64 but it doesn't work, can you please point me in the right direction?
In fact, adding G0Z0 crashes the program when I click on a gcode line in updateRaster() which doesn't happen if the line is not added...

I am new to qt so if you can describe the flow to me, I can try fixing it myself.

Dante-666, 2017-11-13:
I found the problem, I was using Qt 5.9 and it doesn't support QGlWidget that well, I tried changing the implementation to QOpenGlWidget and it works excellent, thought it breaks the QFileDialog that opens when "Open" action is clicked, that can be fixed by using the flag DontUseNativeDialog

Are you planning to migrate to the latest version soon?

mrmimo, 2018-01-07:
I confirmed Dante-666 problem in Fedora 25 to 27 (Qt 5.9.1). QGlWidget doesn't work at all. QOpenGlWidget works as expected. I also had to disable using native dialogs.

gearoid-murphy, 2021-01-22:
The visualization problem is caused by the default settings not being set correctly. Go into the settings menu and select the 'reset to defaults' button. Be sure to save your settings.ini file prior to this as any custom gcode will be lost.

domdfcoding, 2021-02-05:
@gearoid-murphy's suggestion worked for me on Ubuntu 18.04, although I had no issues at all on Ubuntu 20.04

fts-tmassey, 2024-10-15:
Hello!

I'm trying to use Candle 1.2.15b_x64 on an ancient notebook running Windows 10 2022H2 with an Intel 4 Series Express Chipset (Microsoft WDDM 1.1).  When I open a gcode file, I get no preview in the Visualizer window.  I'm certain the hardware 3D support is lacking.

I found this thread and tried the software attached to reply https://github.com/Denvi/Candle/issues/64#issuecomment-244561769 .  It *does* show the preview properly!

Unfortunately, that is a very old version at this point.  I would *really* love to use the current version, but take advantage of the software rendering.  Is there a current version that has been compiled with software rendering?  Could it be possible to just make this part of the normal package, such that you can use an option in the Visualizer settings section to enable software rendering?  It might also help for debugging when there are 3D problems even for users not using an ancient notebook!  :)

Thank you very much for any help you might be able to provide.  Candle is a *wonderful* tool for beginning users of CNC equipment like myself!  I'm really enjoying it so far.  I really appreciate it.