Denvi/Candle #65: visualization problems

aix357, 2016-08-18:
The state of the instrument and the coordinates are not redrawn when moving it:
[image]

P.S.
How to configure motion axis settings? For example, the number of steps per unit.

Thank you!

Denvi, 2016-08-19:
Hi, '$$' - command output please.

aix357, 2016-08-19:
Output:
_[CTRL+X] < Grbl 0.9j ['$' for help]
$$ < $0=10 (step pulse, usec)
$1=26 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=224 (dir port invert mask:11100000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=192 (status report mask:11000000)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.001 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok_

speak Russian?

Denvi, 2016-08-19:
> $10=192 (status report mask:11000000)

| Report Type | Value |
|:-------------:|:----:|
| Machine Position | 1 |
| Work Position | 2 |
| Planner Buffer | 4 |
| RX Buffer | 8 |
| Limit Pins | 16 |

from https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

Try set mask to 3:

~~~
$10=3
~~~

aix357, 2016-08-22:
I'm set mask to 255. It's work. Thank you!

closed 2016-09-11 by Denvi