Denvi/Candle #68: Keyboard Cotrol

ArcAIN6, 2016-09-04:
When checking the box for "keyboard control"
one would assume the ability to use arrow keys to jog the machine, however, this is not the case. As i cannot find any documentation on what this checkbox actually does, i'm asking here. 

What exactly does this checkbox do?

Also, i've noticed something odd with the heightmap system. 
I start off by putting the machine in position, with the tip of the tool touching the surface of a PCB, then i set x0, y0x z0 for the work coordinates.
Once i start the probing process, it goes through and probes quite nicely, 
then i exit "edit mode" (wish it would automatically leave "edit mode" ) 
at this point, all seems well, i have a height map, however, after returning the machine to x0 ,y0, z0 when i click "send" the spindle starts, the machine begins moving, and the problems begin. 

Issue 1) After creating heightmap, and exiting "edit mode" and returning the tool to work positions 0,0,0, hitting send with start the program per usual, however, the tool never dips into the work piece.. i.e. i have the g-code set to -0.05mm (thickness of the copper foil on the PCB) one would assume, that if starting at 0, and the heightmap offsets by -0.02, that the resulting plunge of the z-axis would be -0.07, however, this isn't the case.. wen i start the send, the machine moves, but the tool is about 5 - 6mm above the work piece. This means no cutting gets done.. (the "check mode" box is NOT checked)

issue 2) In some cases, hitting the "abort" button will send the machine flying towards machine 0 ,0 ,0. This doesn't seem healthy, as it may lead to the spindle, or tool crashing into various things such as work piece, or fixtures.  While i've only had this happen to me once, it did require me to replace the x-axis limit switch, as the crash into the switch had enough force to completely destroy the switch.  (75lb spindle+gantry + 5000mm/sec == big bang!)

In the case of the second issue, i cannot rule out that as a one-off. My guess is either the software sent the "home" command, or the abort button in the software was perfectly timed during a movement, and it allowed the machine to continue moving in the x direction until it crashed.. Your guess is as good as mine, that's why i placed it second, meaning less priority unless it's confirmed elsewhere that it's an issue. 

As for the first issue, i'm not sure what's going on, the numbers simply don't add up... if i start at 0 on Z, and the G-Code is set to make z go down to -0.05, and the height map has 0.078 set for that location, then the z should go to  0.128, however, it appeared to go to 0.297, which is quite different than the expected result.  After looking a bit closer, it appears it is compairing the heightmap against machine coordinates, not work coordinates.

Denvi, 2016-09-04:
Hello.

> keyboard control

Works with Numpad arrows, bindings here:  https://github.com/Denvi/grblControl/issues/62#issuecomment-236342938
Implemented as to be able to use a separate Numpad unit.

> replace the x-axis limit switch, as the crash into the switch had enough force to completely destroy the switch

"$$" - command output please.

For the issue 1 I need have full process step by step with screenshots.

ArcAIN6, 2016-09-04:
unfortunately, i didn't save the console from the crash (hindsight being 20/20 and all that)

I'm currently probing, so i'll create some screen-shots and take a couple of snapshots of the tool position for you once i attempt to mill this board.  (be a couple of mintutes)

ArcAIN6, 2016-09-04:
Here's the images: 

Heightmap Created
https://s18.postimg.org/u5o8laoqh/Heightmap_complete.png

Ready to send
https://s18.postimg.org/hgty1mim1/ready_to_send.png

Sending
https://s18.postimg.org/53h3upsxl/Sending.png

Picture of tool offset
https://s18.postimg.org/dj6oc7vsp/IMG_1494.jpg

----------------------------------
GRBL Settings:
----------------------------------

$$ < $0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=4 (dir port invert mask:00000100)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=255 (status report mask:11111111)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=150.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=20 (homing debounce, msec)
$27=5.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=101.560 (z, step/mm)
$110=8000.000 (x max rate, mm/min)
$111=8000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=200.000 (x accel, mm/sec^2)
$121=200.000 (y accel, mm/sec^2)
$122=25.000 (z accel, mm/sec^2)
$130=625.000 (x max travel, mm)
$131=625.000 (y max travel, mm)
$132=270.000 (z max travel, mm)
ok

ArcAIN6, 2016-09-04:
As for the Keyboard thing..  YAY!
:D

Denvi, 2016-09-04:
Ok, let's look on pic:

[image]

Approx, in the working position with (x; y) = (4.6; 20.275) as on "Sending" screenshot we have heightmap offset with +0.148 mm. Working "Z" in this pos is 0.108 mm. So, tool should be in -0.040 mm from plate surface. Heightmap algorithm works correctly.

Check whether the tool touches board in coordinates (0, 0, 0) before sending program. The tool is rather strong fixed in a spindle? You use one tool for surface scan & milling?

ArcAIN6, 2016-09-04:
The tool indeed touches at 0,0,0, the tool is firmly affixed in the collet, and collet nut is tightened. 

I've checked to make sure i'm not missing steps, on the z-axis, and even when putting almost my full weight on the spindle, the z axis still moves without losing steps.. 

What i know, is the z-axis starts off by touching the piece, at this point i set the z-axis to 0. Once i make my heightmap, i return to x0, y0, z0. 
At this point, i exit "editing" mode, and click the checkbox to enable the heightmap. 
I then hit "send"
The G-Code has the spindle raise 2mm, then jog to the first position, and lower to -0.05mm (the depth for clearing through the copper foil)

This all appears to function fine up until the point of lowering into the workpiece. I'm not sure why, but the tool simply isn't lowering into position correctly.  I've made several attempts, and in each attempt, the same result. the tool ends up just above the work piece, and not actually cutting into it. 

Without the heightmap however, it does cut properly (but a bit deep on one end due to the unevenness of the fixture bed) You can actually see the previous cut in the last photo (no heightmap involved)

I'll keep fiddling around with it until i figure out what's causing it, and if i find the culprit, i'll report back.

closed 2016-11-20 by Denvi