Denvi/Candle #69: Релиз

zagranichnyy, 2016-09-07:
Denvi, когда ожидается новый релиз, какие изменения ждать.

Denvi, 2016-09-08:
В ближайшее время только мелкие исправления.
Какие бы вам изменения хотелось увидеть?

bhgv, 2016-09-10:
- плагинную систему. но плагины чтоб можно было писать на скриптах. я бы предложил луа.
- отказ от дизайна все в одном окне. имхо лучше расположить в закладках типа bCNC
- интерфейс тоже строить скриптом. это и поможет в разбитии на плагины и упростит жизнь в дальнейшем.
- переложить часть работы со контролером на скрипты (например, инициализацию, отправку команд, периодические опросы) это позволит легко добавлять поддержку других контроллеров/прошивок/интерфейсов

кроме того, отказ от дизайна все в одном заметно облегчает отладку, конфигурирование, обновление.

сразу предлагаю вам помощь.

Denvi, 2016-09-10:
Мысли и направление то верное, но вы говорите совсем про другую программу. 
Данный проект задумывался как легкий и удобный инструмент для GRBL прошивки.

selenur, 2016-09-10:
При запуске программы, добавь установку скорости G1 Fххх, хотя-бы равную 100, т.к. попытка ручного управления, или выполнение файла, где нет первоначальной установки скорости, некоторых пользователей вводит в ступор.

bhgv, 2016-09-10:
хм. не за, не против, просто соображения.

- проект можно и раздвоить. причем, ничего не переписывая. скриптодвиг это просто добавка к тому что есть.
- луа полностью встраивается в программу добавляя максимум 100кб. при этом язык и вообще система просто шикарные. 
- в минимальном варианте это просто читатель конфиг файлов. у вас ведь используются конфиги во внешнем файле?
- тяжелее он не станет, qt либы занимают намного больше места, медленней тоже - луа имеет 1 из самых быстрых вм. по портабельности - луа портирован даже на мк.
- удобнее? возможно. как минимум, можно будет подстроить под себя, реализовать макросы, пользовательские кнопки

- давайте я попробую встроить, а вы посмотрите как оно вам? спрашиваю потому что захочу консультаций автора кода что где. это снимет много лишнего гемора.

кстати, пример того что можно оперативно решать встроив скрипты:
> При запуске программы, добавь установку скорости G1 Fххх, хотя-бы равную 100, т.к. попытка ручного управления, или выполнение файла, где нет первоначальной установки скорости, некоторых пользователей вводит в ступор.

Denvi, 2016-09-11:
@selenur 
> При запуске программы, добавь установку скорости G1 Fххх, хотя-бы равную 100, т.к. попытка ручного управления, или выполнение файла, где нет первоначальной установки скорости, некоторых пользователей вводит в ступор.

Не думаю, что программа может решать за пользователя, какую скорость обработки выбрать, если она не была указана явно.
Пусть уж лучше будет ступор и дальнейшее изучение принципов работы с GRBL, чем убитый инструмент.

Denvi, 2016-09-11:
> давайте я попробую встроить, а вы посмотрите как оно вам? спрашиваю потому что захочу консультаций автора кода что где. это снимет много лишнего гемора.

Хотелось бы увидеть архитектуру программы с системой плагинов.
Что из себя представляет плагин. Какие интерфейсы должна представлять программа для плагина и т. д.

Как насчет Qt Script движка? Есть ли у lua значительные преимущества. Ведь JavaScript сегодня знает чуть ли не каждый.

selenur, 2016-09-11:
При фрезеровке когда траектория очень большая, в вуализаторе такая каша, может подумать о возможности скрывать выполненную траекторию?

Denvi, 2016-09-11:
@selenur 
>При фрезеровке когда траектория очень большая, в вуализаторе такая каша, может подумать о возможности скрывать выполненную траекторию?

В настройках можно задать цвет выполненной траектории. Меню "Сервис" > "Настройки" > "Цвета" > "Траектория" > "Выполненная".

selenur, 2016-09-14:
[image]
Если G-код выполняется с ошибками, очень хотелось-бы, автоматическое прерывание работы, т.к. сейчас продолжается выполнение только тех строк, где нет ошибок, и траектория получается непредсказуемая.

Denvi, 2016-09-14:
> автоматическое прерывание работы, т.к. 

Принято, будет добавлена опция "Действие при ошибке": "Прервать отправку"/"Игнорировать".

selenur, 2016-09-17:
С кабелем там всё нормально, просто траектория фрезеровки была очень большой, более 4-х часов, и примерно на половине кода КАМ модуль программы fusion360, вставил команду которая grbl не поддерживается, и дальнейшее выполнение траектории началось в непредсказуемых направлениях, в результате выполнение прервал. 
А т.к. пока нет возможности выполнить с нужного места, я выполненную траекторию удалил в ручную, в блокноте, и немного зацепил коды инициализации, из-за чего такие ошибки.

bhgv, 2016-09-17:
@selenur
рекомендую vectric aspire. очень простой и удобный кам процессор. выходная машина грбл вроде есть из коробки. у меня есть, но из коробки или нет не помню.

дада, есть такая проблема. после сбоя или все сначала или вручную в блокноте. в bCNC есть простенький векторный редактор. можно удалить часть треков, можно изменить часть треков, повернуть, подвинуть. если б не тормоза бо все-на-питоне был бы идеалом.

куда развивать? - в сторону бцнц, но все что тормозит должно быть на с(++). и не питон. не смотря на всю красоту и мощь питон громоздок и не очень быстрый

selenur, 2016-09-18:
Вектрик кроме того что платный,так ещё и поддерживает лишь 2.5d.
В отличие от fusion360, где полноценный 3d, поддержка генерации g-кода для 3-х, 4-х осей, генерация кода для 3д принтера, и много другого, и бесплатно. 
Пост процецессор я подкорректирую, и g-код будет нормальным.
Так что все задачи по генерации g-кода, я перекрываю фьюженом, и своей программой (текст, plt траектории, рисунки).
для просмотра и редактирования g кода, использую nc-corrector 4. В нем большие траектории не так тормозят.

selenur, 2016-09-18:
В будущем было-бы хорошо добавить поддержку лазерного выжигания, т.к. сейчас это невозможно сделать в этой программе, т.к. 3д просмотр нужно делать абсолютно иначе. В связи с чем знакомый сейчас делает альтернативный сендер для выжигания рисунков.

selenur, 2016-09-21:
Есть такая команда $C которая включает режим проверки g-кода, в этом режиме при посылке "?" получаем такую строку: <Check,MPos:5.529,0.560,7.000,WPos:1.529,-5.440,-0.000>, что означает активирован режим "симуляции", при котором парсер G-кода не отправляет команды на драйвера моторов, и шпиндель. Но отвечает "ok" если команда нормально распознана, и "error ..." если что не понравилось. Повторная посылка $C отключает этот режим. 
В итоге это позволяет, если необходимо, предварительно проверить наличие возможных проблем при выполнении G-кода.

Denvi, 2016-09-21:
> 3д просмотр нужно делать абсолютно иначе

Мощность лазера определяется z-координатой?

> Есть такая команда $C 

Эта команда используется флагом "Проверка" в программе, но, при возникновении ошибок, сообщений не выводится. Приходится вручную пролистывать таблицу со списком команд и смотреть на ответы.

selenur, 2016-09-21:
В моем случае при использовании лазера Z-координата не используется, а используется один из вариантов:
- PWM выход, с постоянным периодом включения.
- Задержка в точке без изменения мощности,на разное время в зависимости от необходимого цвета.

Про 3д, и иначе, я имел ввиду следующее:
Например рисунок размером 10х10 см, с размером точки 0.1 мм, может содержать до  100000 точек, в связи с чем отображать громадные массивы точек, в OpenGL становиться не рационально, из-за падения производительности, оптимальнее после выбора файла с g-кодом получить рисунок будущей траектории, и уже его отображать в OpenGL.

Denvi, 2016-09-21:
> в OpenGL становиться не рационально

Для этого была введена опция "Упростить геометрию" в группе "Визуализатор" настроек программы.
Длина отрезка определяет минимальное допустимое расстояние между двумя точками.

selenur, 2016-09-21:
Весь код для выжигания состоит из команд: перейти в точку, включить лазер с нужной мощностью, либо задержкой, выключить лазер, перейти на новую точку, и т.д. Но к сожалению проверить твой вариант, пока невозможно, т.к. при загрузке файла, программа завершается с ошибкой.

Denvi, 2016-09-21:
Файл нет возможности передать на проверку?

selenur, 2016-09-22:
Конечно есть, напиши мне на zheigurov @ gmail.com  и в ответ вышлю

Denvi, 2016-09-23:
Да, для файлов построенных по типу растровых, а не векторных, необходимо реализовать другой алгоритм визуализации.
Зная исходное разрешение, т.е. количество точек на милиметр, можно создать изображение в памяти на основе g-code программы и использовать его как текстуру в сцене OpenGL.

Проблема с вылетом при загрузке больших файлов временно решаема, но, в перспективе, нужно делать поточную загрузку файла небольшими частями.

bhgv, 2016-09-24:
прошу прощения за задержку
- за луа и qt - вот какието ссылки про интеграцию луа в qt creator
http://www.qtcentre.org/threads/62864-SOLVED-Lua-into-QTCreator-IDE-for-the-basic-windows-8-user
http://essgeeeich.github.io/
есть еще проект qtlua (или luaqt?) легко гуглится. еще есть биндинг в с++, только помоему это лишнее.
но лично по мне все проще и банальнее. для использования достаточно пары инклуд и прилинковать 1 с_call библиотеку.

- насчет перерастания композитных проектов в монолитные бинари. честно говоря, досе наблюдал только обратный процесс. включая участие в нем.
вам ведь не нужна скорость в отрисовке копочек или реакции на них? а там где нужна скорость - бинарь, да. например, много ли вы знаете болемене крупных программ, которые не имеют скриптов или плугинов?

- насчет планирования и архитектуры. окончательной. сердцем стает скриптомотор. скрипты раскиданы по папкам как это делается в питоне. отдельно, папка или папки по разным контроллерам, отдельно по разным парсерам (3д немного отличается), отдельно по утилитам (например, утилита периодического опроса и регуляции тепературы 3д принтера), гуй тоже в отдельной папке. или по папке на каждое окошко или функциональную группу.
отдельно папка для плугинов, отдельно для бинарных либ.
будет много папок и много маленьких файлов. и это очень упростит работу и отладку. и никак не скажется на работе программы.

бинарный парсер гкода подключаемый к луа могу предложить уже сейчас (к луа несложно пишутся С расширения)

вот, хорошая книжка - https://www.lua.org/pil/contents.html . про старую версию, но изменений не так много и они не принципиальны.
про встраивание с главы "Part IV · The C API"

разные туториалы
http://lua-users.org/wiki/TutorialDirectory

редакторы/иде
https://studio.zerobrane.com/
http://wxlua.sourceforge.net/

еще луа понимают сцинтилла/скайте и все на ней основанное, нотепад++ и в принципе, большинство прогредакторов.

попробуйте ради интереса собрать https://www.lua.org/source/5.3/ это очень просто.

bhgv, 2016-09-24:
в целом, может я даже сделаю макет и покажу. архитектура обычная. в центре скрипт, а все остальные части, как бинарные, так и скриптовые - плугины. например, папка - машины. в ней папки - грбл, тиниг, тикап, репетир. в каждой из них скрипты - инит, фини, сенд, еррор итд. в таком стиле. гуй и прочее также. и никакой путаницы

Denvi, 2016-09-24:
> луа и qt - вот какието ссылки про интеграцию луа в qt creator ...

Спасибо, посмотрю.

> в центре скрипт

Имеется ввиду скриптовый движок (интерпретатор)?

bhgv, 2016-09-25:
@Denvi 
полностью безскриптовых.. возможно, есть такие. но скрипты очень облегчают конфигурирование, пользовательские функции и прочие вещи не ьребующие высокой скорости обработки.

сперва стартует интерпретатор, и по скриптам подгружает нужные модули, бинарные или скриптовые, неважно. строит интерфейс, учитывая, например, пользовательские установки: просто сендер,  сендер с редактором, сендер под лазер, под 3д принтер итд. грузится обвязка под текущий контроллер станка. они на 90% одинаковые, но не полностью.

плагины можно писать и на луа и на с/с++. луа умеет грузить обычные длл/со. токо интерфейс описать. это просто. статья:
https://www.lua.org/pil/26.2.html

тоесть в бинарной со/длл либе "myDll.so" при загрузке из луа первой будет вызвана функция "int luaopen_myDll (lua_State *L);
вида:
```C
 int luaopen_myDll (lua_State *L) {
      luaL_openlib(L, "myDll", myDll, 0);
      return 1; //это число результатов возвращаемых в интерпретатор
 }
```
интерфейсы с-луа этого модуля тут передаются через список myDll вида:
```C
struct luaL_reg myDll [] = {
      {"название функции или переменной для луа", название_соотв_С_функции},
....
....
      {NULL, NULL}  // конец списка
    };
```
можно и по другому.
можно внедрять С данные в интерпретатор. ссылки, куски памяти.
можно подключать С данные, функции, модули к луа сборщику мусора.

вобщем, скрипт это +. особенно, если не писать абсолютно все на скрипте и не писать абсолютно все на С.

Denvi, 2016-10-04:
Спасибо за информацию. Буду изучать.

bhgv, 2016-10-05:
днями набросал макет
https://github.com/bhgv/lua.gcode.sender

что работает (за образец интерфейса брался bCNC. иконки оттуда же):
- загрузка гкода (вкладка "файл")
- отрисовка гкода
- вывод гкода на ком порт (вкладка "контрол")
- просмотр выводимого кода/ответов контроллера в терминале (вкладка "терминал")

остальное пока не реализовано или не подключено

все скрипты кроме **bld/init.lua** находятся в папке **bld/conf** и разбиты по функциям/вкладкам интерфейса.

парсер (**gcode-parser**) написан на **coco/r** (C++/unicode форк). точнее, переделан из питон варианта. артефакты присутствуют.
сам сосо включен **gcode-parser/tools/coco_cpp**, но скомпилирован под линукс-арм. для других систем-процов надо пересобрать

окошки и потоки - tekUI,
ком интерфейс - lua-periphery.
ссылки на исходные репо в ридми

Denvi, 2016-10-05:
Будем посмотреть.

Denvi, 2016-10-15:
Решил пока остановиться на плагинах Qt, скрипты под большим вопросом, скрипт в основе - однозначно нет. Сложно для меня собирать приложение из такого большого набора современных технологий...

bhgv, 2016-10-31:
плагины так плагины. дорога в 1000 ли начинается с одного шага.

давно не отвечал бо решил довести мой скрипто-сендер макет до чегото более понятного.

счас работает 
визуализатор и в плоскости и в ортогональной проекции, 
редактор г-кода, 
исполнение любой строки или с любой до любой строки. 
указывается номер строки с ошибкой, ее можно исправить в редакторе и продолжить выполнение с нее.
терминал + возможность отредактировать и переслать команду с терминала. просто ткнуть в нее мышкой
контроллеро-зависимые части вынесены в отдельный модуль.
парсер г-кода написан на парсеро-генераторе. и его тоже несложно дополнить/изменить.

много конечно и недоделок. например, не сохраняется измененный г-код, даже не начинал плугинный интерфейс, но это когда закончится основа. визуализатор пока не идеален.

но, по моему, как минимум, хороший пример что можно сделать за месяц разбив на модули и используя скрипты для интерфейса и вспомогательной логики и некритичных к скорости модулей.

https://github.com/bhgv/lua.gcode.sender

bhgv, 2016-11-06:
@Denvi 
прошла неделя. 
- написал двигатель для плугинов "conf/utils/plugin_system_engine.lua" (автоподхват из папки плугинов, отдельный поток и глобальная среда для каждого, связь с гуи для конфигурирования, перехват ошибок итд).
- написал 3 плугина (conf/plugins): 
--- трасер картинок в gcode (можно применять и для карт высот), 
--- создатель шестеренок, 
-- создатель профилей лопастей для roots pump. редкость, кстати. не нашел в бесплатном доступе. 
код получился очень простой.

все, включая скомпилированые бинари, бинарные модули, скрипты, шрифты и картинки (которые, видимо, не понадобятся и их можно удалить) весят 4мб.

разве может быть лучший довод в пользу модульности и использования где надо скриптов?

Denvi, 2016-11-20:
Закрыто до следующего года.

closed 2016-11-20 by Denvi