Denvi/Candle #70: периодический вылет по ошибке. сборка 0.8.4. windows 7. видео интел встроенное

bhgv, 2016-09-10:
вылетает по ошибке каждый раз при закрытии и в разное время при работе (посылке команд).
ошибку попробую отловить, если сборка дебуговая.

кроме того, довольно скоро после начала посылки, опять же в разное время, перестает двигаться карандашик и перестают одновляться текущие координаты. при этом синяя линия на чертеже двигается нормально (дефолтные цвета)

Denvi, 2016-09-10:
Здравствуйте. 
1) Пробовали ли сборку из #64?
2) Попробуйте увеличить период опроса состояния GRBL.

bhgv, 2016-09-10:
только что попробовал 64 и 64.1. как минимум при закрытии вылетают обе при обращении по 14h адресу гдето в ig4icd32.dll

 mov         eax,dword ptr [eax+14h] ; eax == 0

не могли бы вы выложить дебуг сборку актуальной версии для посмотреть где сбоит? на вин машине у меня нет qt, она перегружена и если есть возможность не ставить это будет лучше.

Denvi, 2016-09-12:
Отладочная версия: https://drive.google.com/file/d/0B4vTX7JEeIAYWFp0cnRmeklBOEk/view?usp=sharing

bhgv, 2016-09-12:
#64 идет стабильнее за последний день ни разу не вылетела при работе. хотя, с нераспознанными гкодами надо чтото делать. приостанавливаться и показывать этот код в редакторе чтоб поправить и продолжить или просто продолжить?

еще интересна возможность запускать гкод с произвольного места, скажем, с курсора в редакторе (например, при сбое чтоб не гонять шпиндель вхолостую).

отладочную завтра уже.

bhgv, 2016-09-15:
дебуг версия тоже не вылетает в середине отправки. возможно, дело в оптимизаторе? чтото лишнего удаляет?

Denvi, 2016-09-17:
> возможно, дело в оптимизаторе 

Сложно сказать без локализации ошибки.

bhgv, 2016-09-17:
> Сложно сказать без локализации ошибки. Проблема то в видео-драйвере от Intel. Обновлен?
драйвер последний с сайта леново.  посмотрю более новые, но есть 2 соображения.
1) вылетает только во время отправки команд.
2) дебуговая версия не вылетает при работе уже несколько дней. только при выходе из сендера, что не так важно.

насчет локализации ошибки - случайное время возникновения лично мне напоминает ошибки динамической памяти.
```C 
/*после освобождения память еще некоторое время может быть доступна и не выдавать ошибок доступа, но при этом может быть выделена и изменяться в другом участке кода. может в какойто момент при очередном освобождении если освободится целая страница быть деаллоцирована полностью и уже выдавать ошибку доступа, но программа к тому времени уйдет далеко от места возникновения. */
```
ловить такие ошибки непросто.
для ловли выделения памяти под линем есть волгринд, но под моим арм-линем этого сбоя нет. еще можно разбить программу на куски и запускать в разных комбинациях, особенно удобно если делать из скрипта. не надо перекомпилировать каждый раз.

оптимизация -О2 элиминирует заголовки функций и более агрессивно переменные. может, попробовать -О1? у вас не используеся самоотладка? ей нужны заголовки.

чувствую, я и сам гадаю и вам голову морочу. надо или ставить qt и смотреть самому или успокоиться, что дебуг версия болемене работает.

> Согласен, хорошо бы реализовать.
еще можно с произвольного места и произвольные участки. еще можно доп датчики. например, не остановился ли шпиндель (сбой питания, завяз в материале итд)?
токо делать это уже когда будет скрипт энжин. почему? - потому что скорость, цена и сложность написания-отладки на скрипте на 3 порядка меньше.

почему луа а не жс (лично для меня)? 
- хотя это скрипты примерно одного порядка
- луа очень прост как сам синтаксис, так и его исходник, так и техника его ембеда екстенда.
- достаточно мощен. там есть таблицы и хэши и обьекты с наследованием и перегрузкой операций. итераторы. лямбда функции.
- диктует более строгий и читаемый стиль. хотя, конечно, не питон.
- есть модуль встроенного дебуггера.
- при встраивании можно подцепиться из С к любому типу данных, операции, вызову. причем, делается это просто. 
- весьма неплохой гц. можно указывать области видимости гц, можно запускать гц вручную (например, при простое)
- имеет одну из самых быстрых вм (не классическую стековую). есть жит компилирующий вариант (luajit). причем, сама вм как и весь сорец небольшие и написаны хорошо и читабельно.
- это основной скрипт для игр. те он отлажен тысячами игрушек начиная с 90х. есть и другие применения кроме игр, например в распределенке node9 (спорный момент. лимбо кроме простоты синтаксиса лучше во всем) или в прошивке nodemcu.
- еще 1 предпочтение для меня - опыт. опыт написания скриптов, встраивания, изменения сорцов самого языка. такчто я немного предвзят. жс не люблю изза веба. потому трудно судить.

- широкая известность жс и преподаваемость в школах как + так и -. с одной стороны, это привлечет людей, с другой стороны, сильно либеральный синтаксис простимулирует небрежное, нечитаемое и неотлаживаемое программирование. 
синтаксис луа похож на жс, но имеет небольшой шаг в сторону паскаля и не разрешает безалаберный стиль жс. 
дока по луа изучается за полчаса.

- структура с плугинами:
- вначале это просто замена конфигу (сами плугины будут просто модулями луа. как на луа, так и на с/с++).
- ближе к концу - прослойка между интерфейсом и С(++) обработчиками. это позволит упростить и отладку и добавление функционала по мелочи. например - останавливать или не останавливать по ошибке? какие команды посылать при нач инициализации? алгоритм управления с клавы (или джойстика). итд
- еще ближе к концу - в машкоде остаются только части требующие скорости. например, парсер гкода, визуализатор, еще что будет тормозить.
- опционально, не знаю как оно счас, парсер гкода переписан под парсергенератор. гкод счас развивается. так будет проще дополнять парсер.

сам код программы, особенно поначалу, менять не надо. интерфейс к скрипт энжину это отдельный файл с дергалками уже имеющихся функций

selenur, 2016-09-18:
Поддержка скриптов была-бы не лишняя, т.к. я планирую сделать контроллер на stm32, с таким-же протоколом grbl, только с расширенным функционалом и опциями, и использовать этот сендер, а не переписывать свой, т.к. времени свободно сейчас мало.

Denvi, 2016-09-18:
> луа 

Есть ли примеры интеграции в Qt-проект? Работают ли с ним механизм сигналов/слотов?

> замена конфигу

Боюсь, что конфигов сейчас нет.

> в машкоде остаются только части требующие скорости

Что-то питоном да ктквиком пахнуло. Не очень нравится этот подход, рано или поздно приводит к переписыванию всего и вся на машкод.

> дергалками уже имеющихся функций

При написании программы в текущем виде модульность не предполагалась, так что набор имеющихся функции будет недостаточен.

В целом, с оглядкой на список желаний, таких как: модульность, дополнительные функции, различные режимы (лазер, 3d-принтер, фрезер и т.д.), поддержка различных прошивок, нужно сперва разработать и реализовать подходящую архитектуру ПО. В противном случае будет мучительно больно двигаться дальше.

Denvi, 2016-09-19:
По вопросам, не относящимся к изначально описываемой проблеме, предлагаю переехать в #69.

closed 2016-11-20 by Denvi