Denvi/Candle #72: Install GrblControl in Raspberry Pi 3B.

jvarl, 2016-09-12:
Hello, I think GrblControl is a very good software to control grbl, with features simple and intuitive operation. I installed GrblControl in five PCs with Windows 7-10 (see #55), and on a PC with Linux debian-Jessie, and it worked very well. At the moment I try to compile the sources in raspbian-jessie (QT5/QtCreator) unsuccessful due to the error presents: cannot find QTopenGLWidget. I analyzed the cause and find that there is no libdrm-Intel1 package and do not have a graphical equivalent acceleration driver vc4 in raspbian for the VideoCore IV 3D graphics core. Mr. Denvi, please, if you could clarify me the following: 1) raspbian-jessie has one driver graphics acceleration that could be used? If so, how the file .pro could be modified? 2) If not the first. GrblControl could be compiled sources (using QtCreator) without graphics acceleration support, but run slower?
P.S: I installed two software, bCNC, and Universal Gcode Sender and they work very well in raspbian-jessie. Thank you. JVARL

Denvi, 2016-09-12:
How do you try to compile? Native or cross-compiling?
Qt version?

jvarl, 2016-09-12:
No cross-compile. Just compiling into raspbian. I installed QT5-default, qtcreator libqt5serialport5 and libqt5serialport5-dev in last versions.

Denvi, 2016-09-12:
~~~
apt-cache show qt5-default
~~~

output please.

jvarl, 2016-09-12:
Yes, it is just the way I installed qt5 and qtcreator in raspbian. I much would appreciate if could answer me questions I did above?

jvarl, 2016-09-12:
Package: qt5-default
Source: qtbase-opensource-src
Version: 5.3.2+dfsg-4+rpi1
Architecture: armhf
Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Installed-Size: 96
Depends: qtbase5-dev, qtchooser
Suggests: qt5-qmake, qtbase5-dev-tools
Conflicts: qt4-default
Homepage: http://qt-project.org/
Priority: optional
Section: libs
Filename: pool/main/q/qtbase-opensource-src/qt5-default_5.3.2+dfsg-4+rpi1_armhf.deb
Size: 24706
SHA256: fce82ad22785fbeefc77ae55549f579f7ee4160b6a131c02148c9c4db8a1dd74
SHA1: 89d20baaf27c7e1d83375b3ffb8b47f4a41eaac4
MD5sum: 95e430119dd8dfe9b7b98fcc09fcbbb7
Description: Qt 5 development defaults package
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 This package sets Qt 5 to be the default Qt version to be used when
 using development binaries like qmake. It provides a default
 configuration for qtchooser, but does not prevent alternative Qt
 installations from being used.
Description-md5: 368d5d334866ac98243e1eeb37c32cdc

Package: qtcreator
Version: 3.2.1+dfsg-7+deb8u1
Architecture: armhf
Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Installed-Size: 37691
Depends: libqt5sql5-sqlite, qml-module-qtquick-controls (>= 5.2.1-4~), qml-module-qtquick2 (>= 5.2.1-5~), qtchooser, qtcreator-data (= 3.2.1+dfsg-7+deb8u1), libbotan-1.10-0, libc6 (>= 2.15), libgcc1 (>= 1:4.4.0), libqt5concurrent5 (>= 5.0.2), libqt5core5a (>= 5.3.0), libqt5declarative5 (>= 5.0.2), libqt5designer5 (>= 5.0.2), libqt5designercomponents5 (>= 5.0.2), libqt5gui5 (>= 5.3.0), libqt5help5 (>= 5.0.2), libqt5network5 (>= 5.0.2), libqt5printsupport5 (>= 5.0.2), libqt5qml5 (>= 5.2.0~beta1), libqt5quick5 (>= 5.1.0), libqt5script5 (>= 5.0.2), libqt5sql5 (>= 5.0.2), libqt5webkit5 (>= 5.2.0), libqt5widgets5 (>= 5.2.0), libqt5x11extras5 (>= 5.1.0), libqt5xml5 (>= 5.1.0), libstdc++6 (>= 4.9), libx11-6, qtbase-abi-5-3-2, qtdeclarative-abi-5-3-2, qtquick1-abi-5-2-1
Recommends: gdb, make, qt5-doc, qtbase5-dev-tools, qtcreator-doc, qtdeclarative5-dev-tools, qtquick1-5-dev-tools, qttools5-dev-tools, qttranslations5-l10n, qtxmlpatterns5-dev-tools, xterm | x-terminal-emulator
Suggests: cmake, g++, git, kdelibs5-data, subversion
Homepage: http://qt-project.org/wiki/Category:Tools::QtCreator
Priority: optional
Section: devel
Filename: pool/main/q/qtcreator/qtcreator_3.2.1+dfsg-7+deb8u1_armhf.deb
Size: 9253526
SHA256: 5cd12ef811190005bd806d025eaa54ac0573474989e6619affef92b8730498f1
SHA1: 351f32e0d638aff9486e6ccbb44011d759f4d45a
MD5sum: 185336690ead46dd7c9727e4e5fa301f
Description: lightweight integrated development environment (IDE) for Qt
 Qt Creator is a new, lightweight, cross-platform integrated development
 environment (IDE) designed to make development with the Qt application
 framework even faster and easier.
 .
 It includes:
  * An advanced C++ code editor
  * Integrated GUI layout and forms designer
  * Project and build management tools
  * Integrated, context-sensitive help system
  * Visual debugger
  * Rapid code navigation tools
  * Supports multiple platforms
Description-md5: 7885cd3da2a1e224d06682f5f2a70d2e

Package: libqt5serialport5
Source: qtserialport-opensource-src
Version: 5.3.2-2
Architecture: armhf
Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Installed-Size: 98
Pre-Depends: multiarch-support
Depends: libc6 (>= 2.15), libgcc1 (>= 1:4.4.0), libqt5core5a (>= 5.3.0), libstdc++6 (>= 4.1.1), libudev1 (>= 183)
Multi-Arch: same
Homepage: http://qt-project.org/
Priority: optional
Section: libs
Filename: pool/main/q/qtserialport-opensource-src/libqt5serialport5_5.3.2-2_armhf.deb
Size: 30424
SHA256: 3bab4764408f4dd8c1d953d296baebdaf897be53af369c657e6ab130b6d3072f
SHA1: 7c47254630d7b1564cbeb2913d8f9b91662fc17a
MD5sum: 1a939975348bfd3c06f908e974299599
Description: Qt 5 serial port support
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 This package includes the library to use serial ports from within Qt 5.
Description-md5: 495baaf3e432acf5b4e4136a5d0e645a

Package: libqt5serialport5-dev
Source: qtserialport-opensource-src
Version: 5.3.2-2
Architecture: armhf
Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Installed-Size: 38
Depends: libqt5serialport5 (= 5.3.2-2), qtbase5-dev
Multi-Arch: same
Homepage: http://qt-project.org/
Priority: optional
Section: libdevel
Filename: pool/main/q/qtserialport-opensource-src/libqt5serialport5-dev_5.3.2-2_armhf.deb
Size: 10940
SHA256: f3c4a6eafc1452b6ba987cb481f93af143280cfc071a1877758017739fa9b793
SHA1: 176fdf882fc2893bacca49a731151e88f4b45367
MD5sum: 36f455fe09757a34396052f4b8a1e9ff
Description: Qt 5 serial port development files
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 This package includes the development headers for the serial port library.
Description-md5: ec6e5c3fc91d00795e2fd511b16872a7

Denvi, 2016-09-13:
> Version: 5.3.2

QOpenGLWidget has been added in 5.4 as I know.
Update, upgrade...

jvarl, 2016-09-13:
In official update package do not exists 5.4 version, then it is not possible at this moment. Also is not recommended to use external repo's (only experienced people at risk). We must wait developers introduce this update in the stable version of raspbian-jessie. Meanwhile, I would try to compile GrblControl (current qt5-default/qtcreator 5.3 versions) in raspbian without graphic acceleration support if that were possible. You could tell me how to do this?

Denvi, 2016-09-13:
> if that were possible

I don't think that it is possible as graphics engine is written on OpenGL. 
Besides, I don't see in it any sense in view of low cpu performance of raspberry pi.

jvarl, 2016-09-13:
Esto es al solo efecto de comparar performance de software. Yo no creo que los dos controladores que anteriormente mencioné e instalé, que funcionan bien en raspberry, utilicen aceleración gráfica.

jvarl, 2016-09-13:
This is the sole purpose of comparing performance software. I do not think the two controllers that I mentioned earlier and installed, they work well in raspberry, use graphics acceleration.

jvarl, 2016-09-13:
Not having additional reason for interest for the moment I am about to close the article that I have opened. Much appreciate your respectful attention, and I remain aware any future contribution I can make derived from my tests.

closed 2016-09-13 by jvarl