Denvi/Candle #73: Смена языка/Translations

DrGrbl, 2016-09-18:
Как сменить язык?

Denvi, 2016-09-19:
Язык программы устанавливается автоматически в соответствии с системной настройкой "Язык системы".

DrGrbl, 2016-09-19:
А вручную можна изменить? Например, что бы язык системы был английский, а язык программы русский.

Denvi, 2016-09-19:
> системной настройкой "Язык системы".

Настройкой "Формат" региональных стандартов.

> язык системы был английский, а язык программы русский.

Переименуйте файлы "grblControl_ru.qm" и "qt_ru.qm", находящиеся в папке "translations", в "grblControl_en.qm" и "qt_en.qm" соответственно.

DrGrbl, 2016-09-19:
Спасибо, все работает!
Но, есть, к примеру, g-код (см. вложение), обработка по коду работает, но визуализации нет, не видно траектории! В других программах все показывает.

[Hello.zip](https://github.com/Denvi/grblControl/files/481033/Hello.zip)

Denvi, 2016-09-20:
Описывалось здесь #26.
В вашей траектории не установлена координата Z.

closed 2016-11-20 by Denvi